acados/acados #1: how do we couple with casadi?

ghorn, 2016-04-05:
We want to use casadi codegen and JIT compilation. The strategy at the moment is to write mostly low-level C code, and then write a `casadi::FunctionInternal` wrapper around it.

jaeandersson, 2016-04-05:
Just to point out again: `casadi::FunctionInternal` does not have a stable API and there is no ambition in making it stable. `casadi::Callback` provides a stable API to `casadi::FunctionInternal`. Is there any good reason not to use `casadi::Callback`?

jgillis, 2016-04-05:
As I understand, the only reason is that Callback _currently_ has no way to specify codegenerate instructions. I think that should be added from our part.

jaeandersson, 2016-04-05:
Yeah, but that's trivial to add.

ghorn, 2016-04-05:
Yes I believe it's just codegen missing.

jgillis, 2016-04-05:
Sure; I understood Greg wanted to prototype right now (a matter of hours), rather than days. As soon as we have extended Callback, the work should be rebased on that.

jaeandersson, 2016-04-05:
I'm sure Greg is able to add a callback redirection to e.g. `FunctionInternal::generateBody` in matter of minutes. By implementing the corresponding function in `CallbackInternal` and `Callback` (just make it `#ifndef SWIG` since CodeGenerator is only partially exposed to SWIG).

jaeandersson, 2016-04-05:
Those two functions have one line and two lines of code respectively...

roversch, 2017-05-10:
We chose for a loose coupling with Casadi, basically code generating models and dynamically loading them into `acados`. I'll close this issue for now, if we want tighter coupling to Casadi, e.g. as a plugin, we can reopen the discussion.

closed 2017-05-10 by roversch