acados/acados #2: better input/output declaration in functions

roversch, 2016-07-22:
We should use `const` pointers to `const` values for inputs, and `const` pointers for outputs.

This way, we will have a more 'functional' interface.

jaeandersson, 2016-07-22:
What's the point of const pointers? You're copying the pointers by value anyway, not by reference. In CasADi, we do `const double*` for inputs and `double*` for outputs. I see no advantage in having `const double* const` and `double* const`.

rienq, 2016-07-22:
As I remember it, this extra const declaration would prevent a function from changing the pointer itself to different parts of the memory right? Which I think makes sense in our case..

jaeandersson, 2016-07-22:
It makes sense if your functions are so bloated that you don't know what's going on inside them...

jaeandersson, 2016-07-22:
It's not a big deal, but mutable pointer inputs are useful if you want to copy them to local variables:
```c
void foo(const double* inputs, double* outputs) {
 double a = *inputs++;
 double b = *inputs++;
 double c = *inputs++;
 double x = a+b+c;
 double y = a*b*c;
 *outputs++ = x;
 *outputs++ = y;
}
```

jaeandersson, 2016-07-22:
And it makes sense for vector operations:
```c
void axpy10(double a, const double* x, double* y) {
 for (int i=0; i<10; ++i) *y+++=a**x++;
}
```

You're welcome  ;-)

roversch, 2016-07-22:
Sure, but if I wanted to do something like that, I would do it like

     double a = inputs[0];
     double b = inputs[1];
     double c = inputs[2];

which would result in the same, regardless if the pointer was `const` or not.

But as you said, it's not a big deal.

closed 2017-05-10 by roversch