acados/acados #3: Make a formal difference between different argument types

roversch, 2016-07-22:
We identified 4 different types of arguments into a function:
 - input: this may not be overwritten by the called function
 - output: this will (normally) be overwritten by the called function
 - memory: the caller should not overwrite this between calls (warm/hot-starting)
 - workspace: this is scrap, the caller may overwrite this between calls to the function

rienq, 2016-07-22:
:+1:  I like the names for the arguments as well!

rienq, 2016-10-20:
one question: if we would make this decision of always having these 4 arguments, where do we put solver specific user options? as part of the input struct?

roversch, 2016-10-20:
That's a good question. In the condensing interface, I have `args` instead of `memory`, because the condensing code has no memory (as it is right now at least). 

We could make it a fifth argument, or we could include it in input, I slightly prefer the first, because then we have a clearer distinction between the OCP definition and the solver internals.

jaeandersson, 2016-10-20:
Just noting that this is consistent with the memory model we use in CasADi.

dkouzoup, 2017-01-28:
Ups, didn't know that commits with hash tags refer directly to issues. Sorry for the spam!

Ps: I would also go for 5 arguments, input/ouput/args/memory/workspace (see OOQP interface). And we should stick to a common ordering too before it gets messy.

dkouzoup, 2017-02-01:
Hey all,
I had an interesting conversation with Gianluca yesterday on this issue and I personally think it is about time to decide on a common convention for this topic. 

Our suggestion:

1) Arguments for modules (in this order): input / output / args / memory / workspace
2) Memory, input and output are structs
3) Workspace is just a chunk of allocated memory (not a struct)

Reasons:
If the workspace is a struct (as currently in some modules) then it's not really "scrap" memory. Assume for instance that you are using condensing+qpoases. You could calculate the workspace size (in bytes) required for the two modules, allocate the max of the two instead of both and pass it to both functions sequentially. In that way we do not waste memory but rather reuse the same workspace. If we insist on structs, it would better make sense to merge memory and workspace as why would one module alter the workspace fields of another one in between calls?

@roversch, @rienq, @zanellia, let's vote ;) Also feel free to make any further suggestions!

dkouzoup, 2017-02-01:
another point by Robin: if we want, we could still keep some structure of the workspace inside the module, to which the void pointer will be casted into. So to summarize, each module should have:
- module_create_memory (creates the memory struct)
- module_calculate_workspace_size (calculates required size in bytes)
- module (call the module)
- module_free_memory

rienq, 2017-02-01:
Hmm, I agree with the reasons above to have functionality of asking for the size of the workspace and simply allocating the maximum of what you need.. the structs are however useful to keep the code somewhat readable and to not have to take care of memory alignment yourself etc. I am not sure whether it is as simple as casting the void pointer into the desired struct though?

jaeandersson, 2017-02-01:
> Workspace is just a chunk of allocated memory (not a struct)

Opaque memory structures might be an issue for e.g. cross language interoperability (Fortran?). Might not work for functions executed not on a CPU (e.g. via OpenCL). Not sure of this, but might be worth looking into. In CasADi, we opted for avoiding opaque memory for external functions. Another thing, you might want to have a look at the NASA JPL C coding standard: lars-lab.jpl.nasa.gov/JPL_Coding_Standard_C.pdf (the link times out for me, but maybe it's just temporary)

dkouzoup, 2017-02-02:
Thanks for the feedback guys! Rien you're right I guess, it may not be as easy as a simple cast but we could define an ocp_qp_workspace inside the module and then call a function to set up all the pointers to the provided workspace maybe? Similar to what I do [here](https://github.com/acados/acados/blob/master/acados/utils/allocate_ocp_qp.c#L205) with the allocated memory "primalVars" and the struct qp_out? @jaeandersson would that still be an opaque memory struct as you describe it? I remember I read in one of your threads around here (can't find it now) that in casadi you also pass pointers to pre-allocated memory, isn't that the case?

dkouzoup, 2017-02-03:
@rienq  (@giaf, @roversch) I hacked something together for OOQP to demonstrate what I mean. In the script below I switch between structs and chunk of memory using the OOQP_WORK variable:

https://github.com/dkouzoup/acados/blob/master/experimental/dimitris/test_qp.c#L20

and the cast of the memory to the struct is in this function:

https://github.com/dkouzoup/acados/blob/master/acados/ocp_qp/ocp_qp_ooqp.c#L332

(First I just cast the memory to the workspace to allocate the struct fields and then in the function I set up the pointers). Both interfaces seem to work fine now (still need to test with valgrind that does not work on my mac atm)

jaeandersson, 2017-02-03:
> @jaeandersson would that still be an opaque memory struct as you describe it?  

When I said "opaque memory", I only referred to the use of `void*`. In CasADi we try to avoid `void*` in the public API. It's especially weird to use something like that if you use CasADi from e.g. MATLAB or Python.

> that in casadi you also pass pointers to pre-allocated memory, isn't that the case?

We pass pointers for work vectors, yes. What you call "scrap" above. I wouldn't call that opaque memory though. The work vectors are typed (not `void*`). For the memory, it's a bit more complicated.. The CasADi Function class has a public API and an internal API. The internal API works with `void*` memory objects, which is managed by a memory pool. The public API only works with references to the memory pool. When interfacing external functions, it also works with references to a memory pool, but this is not quite stable yet. I'm not sure you managed to follow my reasoning - haha - but the short answer is that in the public API, it's only non-opaque memory objects (and typed work vectors/scrap).

I'm not telling you to copy our design though. And parts of our design isn't even mature, especially regarding the external C function API.

rienq, 2017-02-04:
The solution of @dkouzoup seems very clean to me! I guess someone should still test whether you can then indeed just allocate the maximum size of the different components for which you would like to share the workspace, without potential memory leeks or other issues..

dkouzoup, 2017-02-04:
Thanks for the feedback @rienq ! I'll test that next week calling hpmpc and ooqp with the same workspace and valgrinding the script. If we go for this solution for the workspace, we should then see what to do with the memory, where we need to allocate the sum instead of the max, and possibly already structured. Would we do a concatenation of them or would we nest them somehow?

dkouzoup, 2017-02-08:
Tested it [here](https://github.com/dkouzoup/acados/blob/master/experimental/dimitris/test_all_qp_solvers.c#L139) and it seems to work fine!

rienq, 2017-02-08:
Cool! What do you want to change about the memory though..? Unlike the workspace, the memory is specific and should be preserved for each solver instance separately right?

dkouzoup, 2017-02-08:
yes sure. We were just thinking how to "group" memories of different submodules. We would probably go for nested structs in that case (i.e., nlp_memory has a field that containts the qp solver memory etc). What do you think?

rienq, 2017-02-08:
Hmm, I agree.. but for Sequential Quadratic Programming in particular, I created a qp_solver struct in order to abstractify completely from which QP solver is being used:
https://github.com/acados/acados/blob/master/acados/ocp_qp/ocp_qp_common.h

The problem at the moment though is that I somehow put this in the NLP workspace while it should surely be in its memory instead ;-)

zanellia, 2017-03-07:
What's the status with memory allocation and passing? Is anyone actively working on this? I ended up doing something slightly related while working with Greg. I am going to put the outcome in my "experimental" folder so you can have a look, if  you like ;)

dkouzoup, 2017-03-07:
Hey Andrea! Have a look at the qpdunes interface if you like, it includes more or less most of the things we decided upon: number/order of arguments, functions for initialization of memory and calculation of workspace size, etc. Any feedback is welcome of course!

zanellia, 2017-03-07:
Oh yeah, sorry, I should have been clearer...I am doing something similar for other modules like ocp data, integrators (greg had his own stuff), a gauss-newton module options, init data and so on and started playing with the nested hierarchy. Your stuff seems pretty neat, maybe I can start generalizing the concept to all the rest? As long as we stick to 

> 1. Arguments for modules (in this order): input / output / args / memory / workspace
> 2. Memory, input and output are structs
> 3. Workspace is just a chunk of allocated memory (not a struct)

and 

> module_create_memory (creates the memory struct)
> module_calculate_workspace_size (calculates required size in bytes)
> module (call the module)
> module_free_memory

everything should work quite well after a couple of round of refinements :)

zanellia, 2017-03-07:
Actually, I do realize that most of the interfaces to modules are more or less already in place. What I was missing, and I implemented, boils down to a high-level module "acados" which has all the functionalities needed to interface acados to third party code

zanellia, 2017-03-07:
One more thing :) I would like to propose to change "module_create_memory" to "module_init" (and maybe "module_free_memory" to "module_deinit" for consistency?). A more general function would allow one to carry out other tasks as setting all the options related to the module (and maybe others?).

zanellia, 2017-03-08:
very very last thing (for now): I would also add one more field "data" such that every module would have some more or less static information about itself and other modules. For example: the "acados" module would contain at least a pointer to each of the modules in the lower level such as integrators, qp solvers, condensing and so on and some additional info such as options or other stuff.

rienq, 2017-03-08:
what is the different between the "acados" module that you propose and the ocp_nlp module for which we already have a proof-of-concept implementation.. in the latter, we point to the integrators, qp solver, condensing etc in the memory which is anyway like the data that you propose, right?

zanellia, 2017-03-08:
mmm, I think for what concerns acados it is indeed pretty much what I had in mind...I implemented something else for Greg as they have they're own "modules" to be integrated plus we wanted to do a couple of things differently. 

What do you think about packaging every model into something like this:

`typedef struct _module_name {

  module_name_data  *data;
  module_name_mem   *mem;
  module_name_work  *work;

  void (*init)(module_name_in*, module_name_out*, module_name_args*);
  void (*deinit)(module_name_in*, module_name_out*, module_name_args*)
  void (*run)(module_name_in*, module_name_out*, module_name_args*, module_name_mem*, module_name_workspace*)

} module_name;`

roversch, 2017-03-08:
Just my two cents here;

- I do not really like 'deinit'. Would 'finalize' or 'destroy' or 'end'
maybe make more sense?
- These 'modules' seem like classes, and I do not fully see the point yet
(but you can tell better 'cause it seems you need it). Instead, I thought
of acados as a bunch of 'functions' instead, with a functional interface,
i.e. call it with exactly the same input, memory and arguments (workspace
can be overwritten), it should always give _exactly_ the same output.
- In your example, what would 'module_name_data' be concretely?

Best regards,
Robin

<https://github.com/acados/acados/issues/3#issuecomment-284926825>, or mute
the thread

zanellia, 2017-03-08:
I do realize I am a bit behind with the discussion, but maybe it is useful anyway :) What I am thinking about is indeed something object oriented ( -oriented, as we do not have objects ). I think it would make the code even easier to understand and use. The init function should set up the structure data and allocate memory. Data would contain pointers to sub-modules used by the module and options/configurations. Ideally everything would be set to some default values unless something is provided by the user and it should check that all the minimum information required to initialize the module is provided.

rienq, 2017-03-08:
I agree with packaging the modules because this is something I also needed for the current implementation of the ocp_nlp module, which is still only a proof of concept, but it relies on the additional structs:
- ocp_qp_solver: https://github.com/acados/acados/blob/master/acados/ocp_qp/ocp_qp_common.h
typedef struct {
    int_t (*fun)(ocp_qp_in*, ocp_qp_out*, void *, void*, void*);
    ocp_qp_in *qp_in;
    ocp_qp_out *qp_out;
    void *args;
    void *mem;
    void *work;
} ocp_qp_solver;

- sim_solver: https://github.com/acados/acados/blob/master/acados/sim/sim_common.h
typedef struct sim_solver_ {
    void (*fun)(const sim_in*, sim_out*, void*, void*);
    sim_in *in;
    sim_out *out;
    void *mem;
    void *work;
} sim_solver;

I believe @zanellia is proposing to simply do this for all modules, right? I totally agree with that.. the argument that this seems like object oriented programming is true in some sense, but that argument holds against the use of all the structs we currently already have.. + the modules are simply packaged in structs in order to have easy access to the function pointer, memory, in and output etc. so we still keep all the nice functional properties of the code anyway, i.e., you get the same output for the same input

The second point, of introducing an additional argument called "data" is unclear and seems redundant to me.. we already have a way of passing pointers to submodules and passing options so maybe we should start by identifying the problems with the current approach where we simply pass these things either in the memory or in the input to the module. 
The "sim_in" struct for the simulation module for example has a field called "opts" for passing integrator-specific options. For the ocp_nlp module, I decided to make the QP solver part of the workspace (because it is specific to an SQP method and should not affect the output) while the integrator is part of the input because it is part of the OCP formulation: https://github.com/acados/acados/blob/master/acados/ocp_nlp/ocp_nlp_common.h

roversch, 2017-03-08:
1. Okay, I understand now. We should watch out not to make those structs
too heavy weight, but if they just have pointers to stuff you would anyway
need (like args and mem), then that's fine. I actually liked that when
implementing the Python interface.

2. For me, it seems a bit weird to put 'sim_solver' in the 'inputs', as it
has memory itself. Anyway, for a purely discrete model, it would not be a
'solver' at all, right? just a function evaluation.. So maybe we could
introduce (yet another) layer of indirection, with sim_solver in the
memory, pointed at by some function pointer? Or is this too far-fetched?

zanellia, 2017-03-08:
> I believe @zanellia is proposing to simply do this for all modules, right? I totally agree with that.. 

yes, exaclty.

> The second point, of introducing an additional argument called "data" is unclear and seems redundant to me.. we already have a way of passing pointers to submodules and passing options so maybe we [...]

Everything could be passed in additional structs, but most of the options might be determined at the beginning (when calling init() for example) and never changed again. Right now the code is quite simple, but later on it might pay off to be able to read and write all the options related to a module (and eventually its submodule) from the module itself...what do you think?

> Okay, I understand now. We should watch out not to make those structs
too heavy weight

The struct would actually just group information that we have to store anyway. It should just be a matter of arranging it into the modules.

>  For me, it seems a bit weird to put 'sim_solver' in the 'inputs', as it
has memory itself. Anyway, for a purely discrete model, it would not be a
'solver' at all, right? just a function evaluation.. 

I agree. plus calling ocp_nlp( ... ) to solve an ocp, one should not directly care about how to simulate the ODE/DAE. The way simulation is carried out must be defined at some point and it should be possible to change it, but I would personally like it better if it was embedded in the module an not directly exposed when calling ocp_nlp. This as well would go in the new field "data"

rienq, 2017-03-08:
Did we not have this discussion on the options of the solvers/modules already at the beginning of this thread, where we proposed to have the 5th argument "args" to make options more explicit? I still do not see how data is any different from that? In Andrea's code a big higher up, there is for example "module_name_data" and "module_name_args" but it is not clear to me what is their connection..

I do not agree with the statement that how the dynamics are simulated is not important for ocp_nlp because all our problem formulations are in discrete time, so which simulation routine you use (or which discretized dynamics) defines the corresponding constraints in the OCP formulation and might have an impact on inequality constraints and objective functions as well in case of continuous output-like formulations.. from this perspective, it makes sense to me to pass a simulation solver as part of an ocp_nlp struct and discretized dynamics would just be a special case of that..
So where would you put the sim_solver then? In the memory of ocp_nlp?

zanellia, 2017-03-08:
I would say, when solving an ocp_nlp, everything matters, but most likely you are not going to change the order of your integrator or the the number of intermediate steps from one call to the next, you might just change x_0, some parameter p and options related to the ocp_nlp. By removing all the rest from ocp_nlp_in, and storing it into "data" or "core", the interface to modules might get a bit lighter and cleaner.

> So where would you put the sim_solver then? In the memory of ocp_nlp?

I would add a pointer to sim_solver in ocp_nlp->core/data. 

Think of ACADO's interface, we did not use to pass an integrator and a QP solver to acado_step( .. ), but just x0, parameters and some additional stuff. We do not code generate stuff here, but most likely, once the memory has been allocated you are not going to be able to change certain things, so I would not put them in module_name_in

rienq, 2017-03-09:
Wait, it is not about whether things matter or whether things are changed or not.. the following classification defines where a field ends up:
- problem formulation --> input
- solver-specific options --> args
- memory to solve problem --> workspace (cold) or memory (warm)
- solution --> output

So what you propose, I feel, is not related to the ocp_nlp module where an integrator clearly is part of the problem formulation and should therefore be specified in the input in one way or another, right?
What you propose and what ACADO codegen provides, is not an ocp_nlp module but rather a higher lever MPC/MHE module which has an ocp_nlp solver under the hood. I personally expected such an interface to arise in Python/MATLAB instead of C but we could indeed consider to have such a higher-level module also in C code to simplify some things maybe, but we should have a discussion then including the other developers and maybe not in this thread, which in my opinion is not related to that..

zanellia, 2017-03-09:
> an integrator clearly is part of the problem formulation and should therefore be specified in the input in one way or another, right?

Yes, but generally speaking these kind of things should be decided when the modules are set up rather then when calling them. ocp_nlp does need an integrator, for example, but once the memory is allocated very few things are going to be changed. 

At the end of the day it is really just about moving args into a field inside the module to make things self-contained and intuitive, but it is really subjective, I guess...

A call to ocp_nlp would look like ocp_nlp(ocp_nlp_in, ocp_nlp_out, ocp_nlp_mem) with

`ocp_nlp_in->x0 = ...;
ocp_nlp_in->p = ...;
ocp_nlp_in->sim_solver = ...;
ocp_nlp_in->qp_solver = ...;
ocp_nlp_in->other_modules = ...;
...
ocp_nlp_args = some_options;'

where the last three arguments will stay unchanged most of the time. Storing everything in the module it would be the same with 

'ocp_nlp_in->x0 = ...;
ocp_nlp_in->p = ...;
ocp_nlp_args = some_options;'

I just like the second one better :) Plus using init(), all the stuff that is not specified by the user is going to be set to default values (hpmpc as qp solver, rk4 as integrator, and so on). 

Anyway, let's wrap this up and maybe we can discuss later in a separate thread.

rienq, 2017-03-09:
Of course, there is a use for higher level interfaces and for default configurations to make our lives easier and I am sure that those are the interfaces that most users (including ourselves) will use most of the time.. but the ocp_nlp is the lowest possible interface to a nonlinear OCP problem solver and we should not try to hide anything in my opinion so the goal is rather flexibility and generality over user friendliness

rienq, 2017-03-09:
What you say could even be part of a "ocp_nlp_solver" struct, similar to the "ocp_qp_solver" struct, including functionality like what you propose to make code more readable and to make our lives easier..

dkouzoup, 2017-03-09:
Hey guys, just skimmed through this quite long discussion and I have only two minor remarks (for the moment): I also don't like the term "deinit", would go for "destroy". I still (sorry Andrea ;) ) do not see the need to introduce yet another struct "data" since all the information that would contain can be either part of the memory or of the arguments (i.e. options). I do like the packaging of the modules, and also the init function that would merge for example the initialize_memory and create_arguments functions that I have for instance in qpdunes interface.

rienq, 2017-03-09:
Wouldn't the packaging of the modules be pretty much what you propose as well? It's just that I prefer not to change the interfaces and keep them as flexible as possible, similar to what they are now.. instead of hiding parts that are likely not to be changed often, which could create limitations in the future

closed 2017-05-10 by roversch