acados/acados #7: Condensing with variable dimensions

rienq, 2016-10-07:
The current implementation of the condensing+qpOASES solver does not yet work with variable state and control dimensions on the different stages (even though you could define different dimensions in the current version of the interface).

roversch, 2016-10-25:
@giaf @dkouzoup Has condensing with changing dimensions ever been implemented before? It should not be too hard, but it's gonna be tedious.

giaf, 2016-10-25:
I did, I have a version in hpmpc implementing the O(N^2) and O(n_x^3) condensing algorithm with changing state and input dimensions.

dkouzoup, 2016-10-26:
Once IFAC deadline is passed, I can assist @gianf with this implementation if that helps to speed up things.

roversch, 2017-04-17:
Also see #9

roversch, 2017-08-30:
Supported by the HPIPM interface

closed 2017-08-30 by roversch