acados/acados #9: condensing => partial condensing

giaf, 2016-10-09:
I have a few comments on the condensing interface.

First of all, I would suggest to substitute condensing with partial (block) condensing. This is because partial condensing is more flexible, as condensing is a special case with N=1. And partial condensing could be coupled also with sparse solvers, instead of only qpOASES.

In this framework, it would make sense to consider both the condensing_in and condensing_out as OCP QPs.

Furthermore, there can be several condensing algorithms (actually there are at least 3 with different complexity), so I would suggest to keep the condensing workspace more generic, as just a pointer to void. This would allow for the use of different condensing / partial condensing algorithms (something similar to the QP solvers).

rienq, 2016-10-09:
I completely agree.. I believe this has come up in discussions before, but I guess someone would have to sit down and code this once. Changing the current version of the condensing code to partial condensing and changing the interface to have an ocp_QP as input and output a different, transformed version of that ocp_QP.. that should be relatively easy I think. A tricky thing is however that also the matrices and vectors, which are needed to expand the QP solution back to the full variable space, should be in the output of the condensing. Another thing which is maybe good to keep in mind, is that it can be very useful to have a feature to do "hot condensing" or "vector condensing" where all QP matrices, based on problem linearizations, are fixed from before but the gradients and constraint residuals can be updated. This would be very useful in a multi-level real-time iteration for example!

Anyway, I agree that it would be very good to have condensing more of a separate module which can be connected with pretty much any QP solver. About the different implementations of condensing, this is indeed important to use the right algorithm in the right context but it is maybe also good to stick to one (ideally, the one which overall performs best) implementation for which we try to iterate to a code implementation we are all happy with.. another question at the moment for example is whether we want to rewrite the current implementation based on blasfeo or we want to keep two different implementations in parallel (which would not be ideal because of increased maintenance)?

giaf, 2016-10-09:
I already have the partial condensing coded in HPMPC, since I coupled it with the IPM solver some time ago. The only part missing is the condensing of general constraints, but it shouldn't be much work.

About expanding the solution back, all that is need is the data from the original OCP, that will still live somewhere, so this shouldn't be a big issue either. In HPMPC I coded a "partial_expansion" routine to get back from the solution of the partially condensed OCP to the solution of the full space OCP (complete with multipliers).

About a routine to condense just the RHS, I was thinking about that as well, but I hadn't the time to code it yet. That's in the TODO list :p 

I think I can take the responsibility of play with partial condensing in ACADOS, just probably not in the next few weeks. I could reuse a lot of stuff from HPMPC, and 'just' make changes to the interfaces.

About BLASFEO, maybe it is better to open another issue to discuss it there?

roversch, 2016-10-25:
I think using `void *` is bad coding practice in this case. A cleaner alternative would be to use `union`. See this [example](http://pastebin.com/K6pc9CuV).

giaf, 2017-02-21:
I've been working on this in the last days. Now general constraints and rhs condensing are up and running in HPMPC.

@dkouzoup about (partial) condensing, at the end I realized it is pointless to support the "condense last stage" for partial condensing: it gives no performance gains, and many solvers (e.g. qpDUNES, FORCES) require the arrival cost to be strictly convex). And it makes everything more complicated.

So I kept partial condensing at it was (i.e. leaving the final stage uncondensed). This means that partial condensing can not be used to do full condensing, since a OCP_QP with N=1 in general has nx[1]!=0.

To perform full condensing (and relative expansion), I implemented a dedicated routine. The output would be a special instance of OCP_QP with N=1 but nx[1]==0.
HPMPC can be coupled with that, so now an IP can be an alternative to the AS in qpOSASES also for fully condensed formulations.

Alternatively, it may work also with an OCP_QP with N=0 and using the states at the last stage to hold the optimization variables (i.e. all the inputs).

I do not know what option is better (maybe the second (provided that current software works with N==0), that shows that this is not a multistage problem, but uses states in place of inputs), we should discuss about that as well, before I start writing the interfaces in ACADOS.

rienq, 2017-02-21:
To me, N=1 with nx[1]=0 seems more like a hack, while N=0 states explicitly that there is no optimal control structure at all such that it is a general dense (or possibly still sparse but without the typical optimal control banded structure) optimization problem with all variables just stacked together (all control inputs and possibly also the initial states in case of MHE for example)

But this is more about taste, since pretty much all QP solvers will be able to solve such problems in one way or another as long as we write the acados interfaces accordingly, or am I missing something?

dkouzoup, 2017-02-21:
I think I agree with @rienq 's  choice! Also @giaf, we were mainly discussing the "condense last state" option in partial condensing mainly to support also full condensing without another interface. Is the new dedicated routine for full condensing another algorithm?

rienq, 2017-02-21:
That's a good point: even if there would be no performance benefit, it would be nice to have an (almost the same) algorithm for full condensing and partial condensing mainly for maintainability and extensibility reasons.. I guess it would be a problem for qpDUNES when the last stage is condensed, because it needs a terminal stage right? However, for HPMPC it could work since the last stage can have control inputs as well or not?

roversch, 2017-04-17:
This seems to be the next thing needed. @giaf any news? :)

roversch, 2017-08-30:
Add an argument (partial_condensing_horizon) to the arguments of qp solvers.

dkouzoup, 2017-09-27:
First we should benchmark how much efficiency we lose by keeping partial condensing and QP solver modules separate

closed 2018-04-25 by dkouzoup