acados/acados #11: Make a macro for the sizes of states and controls in different stages?

roversch, 2016-10-20:
ACADO code generation had the feature that the dimensions of states (NX) and controls (NU) were hardcoded. In acados, we pass arrays `nx[]` with possibly variable dimensions. We should investigate if this incurs a performance loss.

A possible way to have both, is defining a macro

    #ifdef CODE_GENERATION
    #define NX 10
    #define nx(i) (NX)
    #else 
    #define nx(i) (nx[i])
    #endif

giaf, 2016-10-20:
Can you explain a bit more in details the behavior of the macro? Do you want to use this in all function interfaces? And do the header files need to be different in the two cases?

In my experience, if the LA is provided by some high performance library, there is basically no performance loss in using nx[ ].
Unless you still want somehow to employ code generation of LA. In that case it would be great to have a clan way to support both :)

roversch, 2016-10-20:
I agree, first we need to make sure if it pays off.

If so, then this would be an elegant solution (there might be other remedies as well). I think this should go in all of the header files, but they don't need to be different for both cases (#ifdef takes care of that). You just need to define a template (for example ##NX## or whatever) and then fill in the right numeric values at code generation time.

rienq, 2016-10-25:
So today, we did some first benchmarking based on the LU factorization in the lifted collocation integrators (the code is here on the github repo) and it seems that blasfeo's LU factorization outperforms the simple implementation we had in ACADO code generation for basically any dimensions except very small ones in the order of the kernel sizes (so this is about 8x8 matrices or smaller, depending also on your architecture)..

So this is already good news, I would say! However, a simple 3-loop implementation like we have in ACADO code generation would outperform blasfeo with a factor of (more than) 2 for these really small problems which we might not like so much.. but then we figured that blasfeo could 'relatively easily' support also such a triple loop implementation without the panel-wise format and you could just switch to this! The majority of the speedup namely comes from not having to transform the matrices into the panel-wise format (which I include in my comparisons since I always transform back and forth). 

We would indeed miss out on like a +- 30% potential speedup by fixing the dimensions of all loops and asking the compiler to do optimizations like loop unrolling etc, but I am not sure whether this is really worth it in the end? What do you think?

giaf, 2016-10-26:
I was thinking about a couple of thinks.

First, to get more reliable timings in linux you may want to run this shell command before starting the tests
`echo performance | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`
to set the frequency of all cores to the maximum. This is especially important if timings are very small. To get back to normal governor, just change `performance` with `powersave`.

Then, I just remembered that in the LU factorization in BLASFEO I have not implemented yet the 8x4 kernels for processing the diagonal blocks. Translation: for matrices up to 8x8 it only uses 4x4 kernels, so there may be some improvement for matrices of size larger than 4x4.

Than said, yes now with the new matrix struct framework it would be easy (and convenient) to add a reference triple-loop version directly into BLASFEO, but without the possibility of fixing the loop sizes. It would still be possible to ask the compiler to loop unroll, but (in case the compiler decides it is a good idea to do so) the speed up would be smaller than in the fixed loop size.

roversch, 2016-10-26:
We should look deeper into this (make more accurate measurements etc.), but from this preliminary test it seems that it would indeed be useful.

giaf, 2016-10-27:
I'm working on that now, I should have the routines needed for the integrators example by later today, so we can do further tests

rienq, 2016-10-27:
Wait, the triple loop implementations would definitely be useful for small problem sizes.. but fixing the dimensions not really in my opinion. Or is that what you meant as well? 
A speedup of about 30% is not really that much, considering that you would have to fix the dimensions of the code basically in all routines in order to get this speedup for the complete algorithm and not just for a tiny part of it, which most likely won't be the bottleneck anymore for such small problem dimensions.

roversch, 2017-08-30:
We decided that this is probably useless.

closed 2017-08-30 by roversch