acados/acados #14: Unit tests

roversch, 2016-10-25:
Unit testing is certainly going to be an important part for acados, as we want to be flexible in adding modules. For the moment, I see the following alternatives to be able to start unit testing ASAP.

1. Write unit tests in pure C (e.g. [check](https://libcheck.github.io/check/web/install.html))
2. Write unit tests in Python, and call the C functions with [ctypes](http://python.net/crew/theller/ctypes/)
3. Write a small SWIG wrapper and do unit tests from either Python or Matlab (eventually, we can call the C functions through casadi, see #1 , but it's probably too long of a process to wait for)

Whatever way we choose, we should start with writing tests for the following:
1. Condensing
2. QP solution (although this is actually the burden of the respective solvers)
3. Integrators @rienq what's the best way to approach this?

rienq, 2016-10-25:
I would currently be in favor of having them in C code actually, since it is not fully clear to me yet how the interface to CasADi would work and it might take a while.. on the other hand, it seems tedious and kinda unnecessary to write our own wrappers to Python or Matlab no? Intead of trying to go towards high-level languages, it would be more exciting to see the first embedded system running acados code since this would be our main set of target applications anyway..

Regarding unit tests, should we maybe make a separate folder and rename the current "test" folder into something like "examples".. I guess the example folder would be a crucial source of information for users to understand how to use acados just like with ACADO

rienq, 2016-10-25:
Oh, and I could give it a try to write unit tests for the integrators at some point.. just like for the other solvers, I guess it is all about quantifying or even estimating the accuracy of the results and setting some acceptable tresholds, no?

mvukov, 2016-10-26:
FWIW, testing execution of ACADO examples is __not__ unit testing. Those are very rudimentary end-to-end testing, where we only check that examples don't crash, but does not tell anything about quality of results. There is one unit test in the `tests` folder -- that might be a good starting point for you. On the other hand, my recommendation would be to use Google's [gtest](https://github.com/google/googletest) (and it looks to me more flexible than `check`).

roversch, 2016-10-26:
Thanks for the link! I'll look into that.

roversch, 2017-01-16:
We went for the catch framework ( https://github.com/philsquared/Catch ).

closed 2017-01-16 by roversch