acados/acados #16: YALMIP to FORCES Pro

rienq, 2016-10-27:
Dimitris found out about the following interesting github repository, which we should keep in mind:
https://github.com/embotech/Y2F

They worked on a tool to automatically detect the multi-stage structure from a general (convex) optimization problem, which apparently is not only a very easy-to-use thing but also overall provides considerable speedups in the resulting FORCES solver! (the latter was also confirmed by Dimitris who used it for his tree-structured QP problem)

Maybe this could be an interesting approach to detect the multi-stage structure in the convex/non-convex optimization problems that one might formulate in CasADi when passing them to acados to the plugins we plan to develop..

jaeandersson, 2016-10-27:
Yeah, this is easy to implement. CasADi already does this for the BlockSqp code: https://github.com/casadi/casadi/blob/develop/casadi/interfaces/blocksqp/blocksqp.cpp#L415. The implementation gets even easier if we allow reordering of the variables, just call `casadi::Sparsity:scc`.

Not sure if @jgillis already added it to hpmpc too (https://github.com/casadi/casadi/issues/1865), but it's something we've talked about.

jgillis, 2016-10-27:
> interfaces/blocksqp/blocksqp.cpp#L415. The implementation gets even
> hpmpc too (casadi/casadi#1865

Did detection, not reordering

roversch, 2017-08-30:
This is a long-term item, since we will not (for now) support acados as a plugin to CasADi.

closed 2024-05-06 by sandmaennchen