acados/acados #18: Errors in valgrind output of test_hpmpc

zanellia, 2016-11-21:
I have valgrinded test_hpmpc, while debugging some other stuff and the output contains quite a few errors:

`==17290== Memcheck, a memory error detector
==17290== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==17290== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==17290== Command: ./test_hpmpc
==17290== Parent PID: 5679
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x4E5C314: fmax (s_fmax.S:25)
==17290==    by 0x459EA1: d_init_var_mpc_hard_tv (d_aux_ip_hard_lib4.c:149)
==17290==    by 0x44934B: d_ip2_res_mpc_hard_tv (d_ip2_res_hard.c:460)
==17290==    by 0x4122AC: fortran_order_d_ip_ocp_hard_tv (fortran_order_interface.c:576)
==17290==    by 0x4092AB: ocp_qp_hpmpc (ocp_qp_hpmpc.c:131)
==17290==    by 0x4086F8: main (test_ocp_qp_hpmpc.c:521)
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x4E5C314: fmax (s_fmax.S:25)
==17290==    by 0x459EFC: d_init_var_mpc_hard_tv (d_aux_ip_hard_lib4.c:150)
==17290==    by 0x44934B: d_ip2_res_mpc_hard_tv (d_ip2_res_hard.c:460)
==17290==    by 0x4122AC: fortran_order_d_ip_ocp_hard_tv (fortran_order_interface.c:576)
==17290==    by 0x4092AB: ocp_qp_hpmpc (ocp_qp_hpmpc.c:131)
==17290==    by 0x4086F8: main (test_ocp_qp_hpmpc.c:521)
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x51C856: inner_edge_dpotrf_8x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51E335: kernel_dsyrk_dpotrf_nt_l_8x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x51C8B1: inner_edge_dpotrf_8x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51E335: kernel_dsyrk_dpotrf_nt_l_8x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x51C92B: inner_edge_dpotrf_8x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51E335: kernel_dsyrk_dpotrf_nt_l_8x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x51C9C9: inner_edge_dpotrf_8x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51E335: kernel_dsyrk_dpotrf_nt_l_8x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x518BE6: inner_edge_dpotrf_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51A30B: kernel_dsyrk_dpotrf_nt_l_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x5528FBF: ???
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x518C34: inner_edge_dpotrf_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51A30B: kernel_dsyrk_dpotrf_nt_l_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x5528FBF: ???
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x518C98: inner_edge_dpotrf_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51A30B: kernel_dsyrk_dpotrf_nt_l_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x5528FBF: ???
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x518D17: inner_edge_dpotrf_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51A30B: kernel_dsyrk_dpotrf_nt_l_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x5528FBF: ???
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x518BE6: inner_edge_dpotrf_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51A30B: kernel_dsyrk_dpotrf_nt_l_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x5528B3F: ???
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x518C34: inner_edge_dpotrf_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x51A30B: kernel_dsyrk_dpotrf_nt_l_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)
==17290==    by 0x5528B3F: ???
==17290== 
==17290== Conditional jump or move depends on uninitialised value(s)
==17290==    at 0x518C98: inner_edge_dpotrf_4x4_vs_lib4 (in /home/andrea/acados/build/test_hpmpc)`

maybe @giaf knows better?

roversch, 2017-01-16:
Same for `nmpc_example`.. :( By the way, I renamed `test_hpmpc` to `hpmpc_example`, the unit tests should be called 'test'.

roversch, 2017-08-30:
We should check this again, on a problem that uses blasfeo.

dkouzoup, 2018-04-26:
@zanellia can we have an updated log for that? We should investigate at some point where these errors come from (I don't think it's limited to HPMPC atm)

FreyJo, 2020-10-07:
Closing due to inactivity.
Cannot recover what the `test_hpmpc` was.

closed 2020-10-07 by FreyJo