acados/acados #21: Default TARGET for BLASFEO

rienq, 2017-01-06:
Can we change the default target of BLASFEO, at least for the unit tests in acados? Or maybe we should even test multiple (or all) of the targets? @giaf 
I just spent a very long time debugging my ocp_nlp test, while it apparently runs perfectly with TARGET set to GENERIC so there must be a bug in the current version of blasfeo that acados is currently using..

zanellia, 2017-01-06:
Is it possible that you do not have a sandy bridge? What's in your /proc/cpuinfo?

rienq, 2017-01-06:
That would make sense, did the default change then because it worked before? I have core-avx-i I guess, following processor: Intel(R) Core(TM) i7-3720QM CPU @ 2.60GHz

But this is not really the point though, since the tests are also failing on travis at the moment, even though they should not..

zanellia, 2017-01-06:
i7-3720QM is an Ivy Bridge, it should be backward compatible with the Sandy Bridge architecture...I would totally agree on changing the default target and setting up unit testing in blasfeo to avoid these problems. I would not know right away how to do it as you have to test on different machines... @giaf?

zanellia, 2017-01-06:
Below some potentially useful pointers:

Unit testing in OpenBLAS: [https://github.com/xianyi/OpenBLAS/blob/develop/.travis.yml](url)
Architecture detection in OpenBLAS: [https://github.com/xianyi/OpenBLAS/blob/develop/cpuid_x86.c](url)

nielsvd, 2017-01-06:
What about building HPMPC for every platform by default? Using "cat /proc/cpuinfo | grep -m 1 flags" you can directly see the supported instruction sets (for Linux at least).

giaf, 2017-01-06:
Wow, I didn't check my emails for a few hours, and once back I found that :p

I didn't update the BLASFEO version in ACASOS in a very long time. Someone did? Why is it failing now and not before?

It may well be that there is still some bug to be fixed in BLASFEO, it is still very young. Could you find the routine that gives different results with target GENERIC and SANDY-BRIDGE? So I can fix it.

In my opinion, the SANDY_BRIDGE target is a good compromise between performance (e.g. HASWELL is better on recent PCs) and usability (it should work on all reasonably recent PCs). I would be against testing BLASFEO and HPMPC for all possible targets, at least now that they are still in a not-so-stable-yet phase (and note that once I will add other architectures (e.g. ARM, or 32-bit x86), I guess it will be impossible to test all targets on travis). In principle, SANDY_BRIDGE is the trusted target, since I use it all the time on my PCs. 

Alternatively, we could decide to include and build the reference BLAS and LAPACK in ACADOS, and use the BLAS wrapper in BLASFEO. Reference BLAS and LAPACK are supposed to be quite stable. But this means that most of the times they will be used in place of the custom LA routines.

Unit testing for BLASFEO would be rather complicated, and requiring a lot of time to be written. For each LA routine, an unit test should be performed for many combinations of matrices, alpha and beta, and especially for many matrix sizes (since some LA kernels are used only for some sizes), and this for all targets (and not all targets runs on all PCs). I think it would take much longer to set up the unit tests than to write the LA routines themselves. 
For the moment, I tested manually the critical cases for each new LA routine, and it worked rather well IMO (there have been very few bugs even if BLASFEO is very young and very low level).

But if someone is willing to help in setting up unit testing for BLASFEO, we can surely discuss about.

rienq, 2017-01-06:
The reason it is failing now and not before, might simply be that I only wrote the unit test a few days ago.. so it could have been a bug that was there since a long time.. I namely had errors of about 1e-5 and sometimes even 1e-2 of my NLP solution compared to ipopt, while it should have been more like 1e-10 all the time. Then, after days of debugging without having a clue where it could be going wrong, I found out that the GENERIC target works with 1e-10 errors so there must indeed be a bug somewhere in blasfeo..

By having unit tests also for the smaller modules such as QP solutions, linear algebra, integrators, we will be much more efficient in identifying where something went wrong. This said, I understand that we cannot put the complete burden of testing blasfeo only on you.. but it is normal/expected that writing tests might take as much time as writing new code or even more time, but it at least gives guarantees which can save you a lot of time once you get (many) users.

For example, if I have some time to write unit tests for the lifted implicit integrators, then this would also be testing the LU factorizations in blasfeo at least (which is where there is currently a bug in the SANDY_BRIDGE target), the some would hold for some other routines by unit testing hpmpc etc.

I agree that we should let travis at least test separately using the reference and openBLAS, which would fix the issues above..

rienq, 2017-01-06:
This said, if @giaf agrees things are currently stable in blasfeo, it might be time to update the submodule in acados which could already fix the current problem in the chain mass test for OCP_NLP..
Also, is there an (easy) way to change the linear algebra library in order to set up a new build job for travis? maybe @roversch knows?
https://github.com/acados/acados/blob/master/.travis.yml

giaf, 2017-01-06:
I would wait a little before updating either BLASFEO or HPMPC, I'm in the middle of a big change (that is, have the entire HPMPC work with structures from BLASFEO). I can do it when it is time.

Since target GENERIC seems to work, the best fix for the moment may be to use it for the unit tests.

rienq, 2017-01-06:
This is kind of the problem though, I cannot change the default target or library of blasfeo since it is part of the blasfeo code itself (namely in the Makefile.rule).. it would be nice to be able to change these settings from outside, e.g., from acados Cmake build scripts

nielsvd, 2017-01-06:
@rienq: you can specify the desired target as an argument to make. Unless the make-file explicitly overrides your definition (``override TARGET=...``), it will respect yours. Hence, you can select the desired target from the acados CMake file with:
```
make static_library TARGET=...
```

giaf, 2017-01-12:
I made a push with an update of both blasfeo and hpmpc. 

Now I force the recompilation of blasfeo and hpmpc all the time. In theory we also have to recompile everything in acados that uses blasfeo every time that BLASFEO_LA is changed. Any idea how to do that in cmake?

@rienq can you check if the same error in the LU factorization still happens?

jgillis, 2017-01-12:
(Hijacking this thread for a small question) -- how is partial condensing support in this latest update?

roversch, 2017-01-12:
@giaf I'm looking into how to do it the 'proper' CMake way.

nielsvd, 2017-01-12:
@giaf @roversch I don't know whether it exactly does what you want, but maybe consider using the [ExternalProject](https://cmake.org/cmake/help/v3.7/module/ExternalProject.html) module from CMake.

roversch, 2017-01-12:
@nielsvd  Yeah exactly, that's what I'm trying :)

giaf, 2017-01-13:
@jgillis I just made a new commit in acados about that. I just added partial condensing in the fortran interface in hpmpc (not yet as a component in acados, work in progress) when compiled using blasfeo. Note that if there are general constraints on the stages but the last one, partial condensing is disabled right now (again, work in progress).

@rienq All tests run just fine with the update in blasfeo. It was old code failing.

@roversch What is the 'proper' CMake way? 
Please be aware of the facts that
1) the structs d_strmat and d_strvec have different fields for different LAs.
2) some algorithm (as e.g. lifted irk) is implemented differently for different LAs, and compiling for one and linking to another LA version of blasfeo will fail.
Point 1) could be solved by adding dummy fields in the structs definitions, but point 2) can not, and requires to rebuild all code using these structs in both hpmpc and acados if the LA in blasfeo is changed.
A solution could be to take track of what is the LA used in the current acados and hpmpc builds, and recompile everything if it is changed.
Or at least throw an error if trying to use a different LA version in blasfeo.
Any suggestion?

roversch, 2017-01-18:
So it seems CMake already does this for us. Whenever you change `LA` in the CMake configuration files of acados, it will recompile blasfeo _and_ acados. 

This closes the open issues.

closed 2017-01-18 by roversch