acados/acados #22: Restructure unit tests

dkouzoup, 2017-01-13:
Currently, running cmake with the unit tests flag calls both octave (to generate condensing data) and a bunch of ipopt simulations (which requires quite some time). To my opinion, Cmake should only create the necessary makefiles. Then running the makefile should generate the data and compile the tests and finally, running the compiled code should run all tests.

rienq, 2017-01-15:
we also generate with CasADi the symbolic expressions and derivatives though, so the only solution I would see at the moment (without making any of the test files part of the repo) is that we include an extra step in between so you would first call Cmake, then run a test_generation script and then run the makefile based on the unit tests and the generated files but I am not sure whether that would actually be better then..

dkouzoup, 2017-01-15:
hmm, that's true. But can't we just call the "generation script" from the Makefile?

roversch, 2017-01-15:
I was fixing this on Friday, let me commit that tomorrow

roversch, 2017-01-16:
I rewrote the CMakeLists.txt such that:
- during `cmake`, only configuration of the build files is done
- during `make`, generation of Casadi functions, generation of test data and building of all targets is done.

closed 2017-01-16 by roversch