acados/acados #24: unify memory intialization

dkouzoup, 2017-01-16:
We mainly use d_zeros from blasfeo/hpmpc to initialize memory. As the blasfeo headers cannot compile with g++, zeros.cpp is temporarily used for the unit_tests. We should remove this once @giaf adds the necessary code snippet in the blasfeo headers (mainly in the aux ones).

giaf, 2017-04-17:
The headers in BLASFEO are fixed.

dkouzoup, 2017-04-18:
@giaf, can we move the blasfeo commit of acados ahead such that we can use the ext_dep includes? Then we can use d_zeros/int_zeros  throughout the code and include only the two headers.

giaf, 2017-04-18:
I wanted to do that now, but the cmake version in acados has been moved up to 3.2 while in my laptop I have Ubuntu 14.04 with 2.8. I'll do that once back to Freiburg where I have another laptop with Ubuntu 16.04 (and then I'll upgrade this laptop...) 

Why the cmake 3.2 version in acados?

dkouzoup, 2017-04-18:
Not sure. Maybe try to set the minimum version to 2.8 and see if acados compiles? @roversch, are there any missing features in versions lower than 3.2?

roversch, 2017-04-18:
Probably not, but 2.8 is old man :p

Also look at https://www.slideshare.net/DanielPfeifer1/cmake-48475415

giaf, 2017-04-18:
It is not so old, if it is the default version on a still-supported Ubuntu distribution ;)

roversch, 2017-05-10:
Any progress with this?

dkouzoup, 2017-05-10:
I'd give it a bit more time before we start changing this everywhere, since we are still thinking with @giaf how to structure auxiliary functions etc in blasfeo.

roversch, 2017-08-30:
We should consider to remove malloc() and calloc() in favor of d_zeros. This issue is on hold until we allocate all memory up front using calculate_size etc.

roversch, 2017-09-13:
See PR #129

closed 2017-09-27 by dkouzoup