acados/acados #34: Implementation of the an efficient version of the Multi-level RTI

zanellia, 2017-01-24:


zanellia, 2017-04-12:
partially addressed by @rien here:

[https://github.com/zanellia/acados/commit/4dc91d74fc94ad4e1db81a40c8802ff682fb1a38](url)

rienq, 2017-04-12:
Yea, at least in the collocation methods, you can easily freeze the sensitivities now and only do gradient updates (I did that for my CDC paper as a reference in a numerical comparison)

zanellia, 2017-04-12:
That's pretty useful! I think quite some ideas we have discussed about will exploit these kind of features.

roversch, 2017-09-27:
[image] [Multi-level RTI](https://trello.com/c/J61oOGF3/27-multi-level-rti)

zanellia, 2019-05-23:
Hi all, Moritz asked me to bump this since it might be useful for our recent collaborations in projects on systems with very fast dynamics. Ideally, separate preparation and feedback phases should be implemented (easy, I guess) and some kind of way to specify which level of the MLI should be performed needs to be available to the user. 

We also agreed that practical implementation of the MLI on embedded platform might be more involved than implementing the standard RTI (MLI needs some form of parallelism), but it would still be a nice-to-have :)

sandmaennchen, 2024-05-06:
Implemented here: https://github.com/acados/acados/pull/1046

closed 2024-05-06 by sandmaennchen