acados/acados #39: multipliers in ocp_qp_out

dkouzoup, 2017-01-27:
Two questions regarding the dual multiplier fields of ocp_qp_out:
1) are we sure we don't want to distinguish between bounds and polyhedral constraints and go for all of them concatenated (and in an order that we should always look back to)?
2)why don't we merge the multipliers of upper and lower bounds to have nb[k]+nc[k] (btw, there's a typo in the reference.pdf, it says ng[k] instead of nc[k]) instead of 2*nb[k]+2*nc[k]?

giaf, 2017-01-27:
The multipliers are in this form just because it's the output you get form HPMPC, and they have never been used anywhere until now (and that's also where the ng comes from). That was just a stump, now we can discuss about how they should look like ;)

I haven't really got what you mean with point 2). What do you mean by merging the multipliers of upper and lower bound?

dkouzoup, 2017-01-27:
Ha, I thought so ;) I just run into that because I was writing an allocate_ocp_qp_out function. My point in 2) is that assuming that upper lower bounds are not equal to each other (even in this case there shouldn't be a problem I think), only one of the two inequalities can be active and therefore only one of the two multipliers different than zero. So in principle there is no need to store all of them cause half will be zero. Or am I missing something here?

giaf, 2017-01-27:
ok, and then now can you know if it comes from the upper or from the lower bound?

dkouzoup, 2017-01-27:
by evaluating the constraint at the primal solution I guess. I've seen this convention in other software that's why I am bringing it up.

giaf, 2017-01-27:
I see. 

Well, in my opinion it's not such a large memory saving, and it adds extra complexity in recovering the information about the constraint they refer to, when needed.

dkouzoup, 2017-01-27:
Btw, can't we just check the sign of the multiplayer to see if it's of the upper (+) or lower (-) bound?

zanellia, 2017-01-27:
Personally I do not like that format so much, I find it a bit counter-intuitive. Plus, if you solve a relaxed KKT system (as we do for the homotopy-based IP), this convention do not make sense anymore: even if you have upper and lower bounds, both multipliers will be non-zero for the tightened problem. In that case you can still recover the multipliers, but would require some post-processing.

dkouzoup, 2017-01-28:
Okay then, the IP guys have spoken ;) I would like to understand though how both multipliers can be nonzero in the case you mention but let's clarify it better in person. For the record, this comes from the qpOASES user guide:

"The dual solution vector contains exactly one entry per lower/upper bound as well as exactly one entry per lower/upper constraints' bound.  Positive entries correspond to active lower (constraints') bounds, negative entries to active upper (constraints') bounds and a zero entry means that both corresponding (constraints') bounds are inactive."

giaf, 2017-02-16:
I would like to propose a change in their order (following a change that gives good performance gains in HPMPC): go from
[lb ub lg ug]
to
[lb lg ub ug]
Is anyone already using the multipliers anywhere?

rienq, 2017-02-16:
I would be more in favor of explicitly splitting the multipliers for the lower and the upper bounds then, since this would avoid possible confusion in the future about defining the multipliers either separately or in a combined way like how qpoases (and probably qpdunes) does, right?

dkouzoup, 2017-02-16:
I agree with @rienq, that would be consistent with the fields of ocp_qp_in too.

roversch, 2017-08-30:
We decided to go for 

    real_t **lam_lb;
    real_t **lam_ub;
    real_t **lam_lc;
    real_t **lam_uc;

roversch, 2017-09-27:
[image] [Combine multipliers into lamb and lamc. Convention is positive upper bound is active, negative lower bound is active](https://trello.com/c/74Kat3Qo/19-combine-multipliers-into-lamb-and-lamc-convention-is-positive-upper-bound-is-active-negative-lower-bound-is-active)

closed 2018-04-25 by dkouzoup