acados/acados #45: Issue with qpoases status flag 37 in test ocp_nlp

dkouzoup, 2017-01-30:
I noticed that if I run the test_qpsolvers.cpp before test_chain.cpp, then the latter crashes (qpoases returns infeasible first QP).  I valgrinded the test_qpsolvers.cpp code and I see no memory leaks so maybe the problem is in test_chain?

dkouzoup, 2017-01-30:
I also observed that if the test_qpsolvers is placed before test_chain, the only way to avoid the error is to test only ooqp in the first script (not qpoases) and not clear its memory.. Which is extra weird. Any other combination gives error 37 in the next test.

dkouzoup, 2017-01-30:
ping @rienq

rienq, 2017-01-31:
I cannot reproduce this problem on my machine though.. these tests seem both to work without any problem here and this in both possible orders. This said, there are definitely memory leaks in test_chain at the moment since this unit test has not yet been updated with the new functionality to free the allocated memory

dkouzoup, 2017-01-31:
hmm, that's weird. So when you run the unit tests it first prints:

---> TESTING qpOASES with QP: LTI, UNCONSTRAINED
---> PASSED 
---> TESTING OOQP with QP: LTI, UNCONSTRAINED
---> PASSED 
---> TESTING qpOASES with QP: LTV, UNCONSTRAINED
---> PASSED 
---> TESTING OOQP with QP: LTV, UNCONSTRAINED
---> PASSED 
---> TESTING qpOASES with QP: LTI, ONLY_BOUNDS
---> PASSED 
---> TESTING OOQP with QP: LTI, ONLY_BOUNDS
---> PASSED 
---> TESTING qpOASES with QP: LTV, ONLY_BOUNDS
---> PASSED 
---> TESTING OOQP with QP: LTV, ONLY_BOUNDS
---> PASSED 
---> TESTING qpOASES with QP: LTI, ONLY_AFFINE
---> PASSED 
---> TESTING OOQP with QP: LTI, ONLY_AFFINE
---> PASSED 
---> TESTING qpOASES with QP: LTV, ONLY_AFFINE
---> PASSED 
---> TESTING OOQP with QP: LTV, ONLY_AFFINE
---> PASSED 
---> TESTING qpOASES with QP: LTI, CONSTRAINED
---> PASSED 
---> TESTING OOQP with QP: LTI, CONSTRAINED
---> PASSED 
---> TESTING qpOASES with QP: LTV, CONSTRAINED
---> PASSED 
---> TESTING OOQP with QP: LTV, CONSTRAINED
---> PASSED 

and then runs the chain example without problems? Because I can reproduce the same problem on mac, linux and travis atm.

rienq, 2017-01-31:
No idea why, it seems to depend on the compiler as well:
https://travis-ci.org/rienq/acados/builds/197018676

the build failed for gcc-5 even though I am using gcc 5.4.0 myself

roversch, 2017-02-03:
This is not fixed as unit test success depends on order of unit tests. There is an actual bug in `ocp_nlp_gn_sqp.c` that I hopefully fixed, now testing with Travis.

dkouzoup, 2017-02-03:
Weird, I didn't actually close the issue, was this automatically just because I wrote "possibly fixed" in the commit?

roversch, 2017-02-03:
Yeah indeed, that closes it automatically. The bug is now fixed, it was pretty mundane, `ocp_qp_in->Q[N]` was not initialized.

Note that I turned off the tests for HPMPC for now (they suffer from a different bug, on which @giaf is working).

closed 2017-02-03 by roversch

rienq, 2017-02-04:
Oops.. Thanks a lot, @roversch!

dkouzoup, 2017-02-04:
Yeah, thanks for spotting that one! But is this actually pushed? Coz it still doesn't work on my mac

commit 6b1dc8d1 (2017-02-01): possibly fixed #45 with previous commit