acados/acados #47: N+1 controls

dkouzoup, 2017-02-07:
@giaf suggested to introduce N+1 controls (where the last control has always dimension 0) such that we avoid treating the last state separately at the end of the loop in many cases. Parts of the code that are already written with N controls in mind, they should anyway remain unaffected by this change. I wanted to point this out to everyone such that you don't think that [this](https://github.com/acados/acados/blob/master/test/test_utils/read_ocp_qp_in.c#L364) for instance is a typo and to make sure that everyone agrees. We should also update the documentation in this case.

dkouzoup, 2017-02-07:
Mainly for @rienq and @zanellia that were not present when we discussed this.

rienq, 2017-02-07:
Hmm, I would agree as long as the last control does not **need** to have dimension 0 for everything to work (in general).. this could namely be useful for some non-standard formulations, including e.g. output noise variables or slack variables on the last stage

giaf, 2017-02-08:
OK we agreed on going for @rienq 's suggestion.

I will write a test problem with soft constraints on the last stage to little by little test all current solvers.

roversch, 2017-04-17:
I labeled this as 'bug' because as of now, it's done inconsistently throughout the code.

roversch, 2017-08-30:
I checked the entire acados library + examples + swig for errors of this kind.

closed 2017-08-30 by roversch