acados/acados #50: Memory leakage due to allocation and not freeing vars in initialise_qpoases()

doanminhdang, 2017-03-03:
I found a memory leakage problem today in the script file ocp_qp_condensing_qpoases.c, then Robin fixed: 
- Move the creator inside initialise_qpoases() to ocp_qp_condensing_qpoases(), so that the memory allocation can be freed. 
Lines: 334-338, file ocp_qp_condensing_qpoases.c.
To be move to the beginning of ocp_qp_condensing_qpoases()
- And then free the memory for primal_solution and dual_solution at the end of ocp_qp_condensing_qpoases().
Uncomment lines 310-311.

rienq, 2017-03-03:
But if it's fixed, you just have to push the commit and close this issue right?
Slightly (un)related question, could we let Travis run valgrind to detect memory leaks on the test and example files @roversch?

doanminhdang, 2017-03-03:
Hi Rien,

He fixed on my computer and I did join the repo yet, so we think it's easy
to just make an issue.

Dang

PS: And I was rushing to the train back home.

roversch, 2017-03-05:
@Rien: good point, it would be nice to have valgrind running automatically.

Some people seem to run it on Travis :
https://github.com/open-source-parsers/jsoncpp/commit/88184d142bc4a61aa47102345218470873d1de8d

The most difficult part will probably be to automate it, as now, I usually
look manually through the output of valgrind.
The valgrind help function seems to offer this possibility:

    --errors-for-leak-kinds=kind1,kind2,..  which leak kinds are errors?

                                            [definite,possible]

        where kind is one of:

          definite indirect possible reachable all none

dkouzoup, 2017-03-09:
I think it's done in casadi too:

https://travis-ci.org/casadi/casadi/jobs/177739247

(first build: TESTMODE = valgrind)

rienq, 2017-03-09:
Talking about travis, it should probably also run all examples to at least check if they run without segfaults and memory leaks.. oder?

roversch, 2017-03-09:
Totally! As valgrind is a bit more usable on Linux than on Mac I did not
look into this yet..

commit 351fbc52 (2017-03-13): Close #50
commit 4c3f79b8 (2017-03-13): Merge pull request #51 from roversch/master

Close #50
closed 2017-03-13 by roversch