acados/acados #52: Breaking blessed/master

zanellia, 2017-03-13:
I mixed up my remotes and ended up braking blessed/master (I reverted everything and pushed to a separate branch. As soon as Travis says it is OK I will push again to master)...Shouldn't we force people to go through a merge such that changes run through Travis before ending up in the master branch? I guess this simply boils down to denying push rights to everybody.

roversch, 2017-03-13:
It seems from [this SO post](http://stackoverflow.com/questions/19734148/restricting-rights-on-github#19734387) that that's not possible.

If you know a way, I would be glad to change the settings accordingly.

mvukov, 2017-03-13:
https://help.github.com/articles/about-protected-branches/

roversch, 2017-03-13:
Wait a moment, the `master` branch is already protected..

roversch, 2017-04-17:
Closing this, as there is no easy solution.

closed 2017-04-17 by roversch

roversch, 2017-05-17:
FYI, Gitlab has this possibility.