acados/acados #55: Update the interface of the integrators based on #3

dkouzoup, 2017-03-23:
We should gradually update all modules to the standards we discussed on #3, regarding type and ordering of arguments.

rienq, 2017-03-23:
Depending on how urgent it is, please wait with this until I cleaned up and merged [this branch](https://github.com/rienq/acados/tree/pseudoSpectral), including inexact Newton implementations of implicit Runge-Kutta schemes of any order, with our master branch

dkouzoup, 2017-04-07:
I guess this is merged, right? I am currently working on cleaning up the nlp_solver interface. When done we can start messing with the integrator interface again ;)

roversch, 2017-04-17:
@rienq any progress with this?

rienq, 2017-04-17:
I will try to do this in the next day(s).. Just to repeat once more, the interface should be:

> 1. Arguments for modules (in this order): input / output / args / memory / workspace
> 2. Memory, input and output are structs
> 3. Workspace is just a chunk of allocated memory (not a struct)

and 

> module_create_memory (creates the memory struct)
> module_calculate_workspace_size (calculates required size in bytes)
> module (call the module)
> module_free_memory

roversch, 2017-04-17:
The qpDUNES interface additionally has the following:

1. `ocp_qp_qpdunes_create_arguments`
2. `ocp_qp_qpdunes_initialize`
3. `ocp_qp_qpdunes_destroy`

The last two are basically wrappers that call `create_memory` and `create_arguments` and the corresponding `free_` functions.

dkouzoup, 2017-04-18:
Yes @rienq, this would be the structure we need. I also use a cast_workspace function (static) inside the module to transform this chunk of memory back to a struct.

dkouzoup, 2017-04-18:
By the way, where will the memory/workspace of the integrators go inside the nlp solver? Also in the input? For the qp solvers, we moved everything to the memory but I am not entirely sure that is the best solution. Would be nice to have a separate discussion about this at some point.

roversch, 2017-04-18:
If you put all of the memory/workspace in one place: the advantage is that it's easier to code (and maintain?), disadvantage is that you might use more space (instead of re-using some of the workspace for example).

closed 2017-05-03 by rienq