acados/acados #56: Proper installation for ExternalProjects

zanellia, 2017-03-24:
Right now ExternalProjects (or some of them at least, like hpmpc and blasfeo) have _INSTALL_COMMAND " "_ in order to avoid cmake runnig install whenever a build is called. However this causes ExternalProjects not being installed properly when _make install_ is called. I bring this up because proper installation might be rather important for integration in more complex build systems and for portability to embedded platforms. I am looking into something like this: [http://stackoverflow.com/questions/37838786/how-to-not-make-install-step-when-building-external-project-with-cmake](url) at the moment. If you happen to have better ideas, please let me know, I actually need right now.

roversch, 2017-03-25:
I like that solution! It's pretty clean.

roversch, 2017-08-30:
Although we want to be modular, it's not really possible to leave out some subprojects, e.g. qpDUNES.

roversch, 2017-08-30:
We also decided to install all subprojects independently, along with a big library including all dependents.

roversch, 2018-02-13:
@zanellia can I close this? The install is done in a modular fashion since some time now.

zanellia, 2018-02-13:
Sure ;)

closed 2019-07-18 by roversch