acados/acados #61: cmake for hpmpc and blasfeo?

zanellia, 2017-04-11:
I had some troubles managing ExternalProjects based on Makefiles from CMake so I made a minimal port of hpmpc's build system to CMake. Not sure this is better than a Makefile-based build system, but at least all our core code is going to be built with CMake (good? dunno). Anyways, if you guys like it we could give it a shot. The code is here:

[I](https://github.com/zanellia/hpmpc/tree/crosscompiling) 

and it is based on a single CMakeLists.txt file (unlike the previous hpmpc's CMake build system)

roversch, 2017-04-11:
Cool! But shouldn't this be an issue in the respective hpmpc and blasfeo
repo's ?

Anyway, this would indeed be very helpful, so please (if @giaf agrees) push
for it.

zanellia, 2017-04-11:
As @giaf is not a big fun of cmake, I thought I could start an acados-based discussion here to convince him :P

roversch, 2017-04-11:
Alles klar ;) We would have the following advantages (to name a few):

   - Compatibility of compiler flags across projects, but you could even
   set different ones if you would want to,
   - Choosing which 'mode' for blasfeo from the top level CMakelists.txt in
   acados,
   - Build targets like make clean etc. would translate more easy to the
   external projects.

Actually, it makes sense to do the same for qpOASES.

zanellia, 2017-04-11:
@ghorn already uses CMake for qpOASES actually...(hint)

doanminhdang, 2017-04-11:
I prefer make, since for some low-level chips the toolchain supports Make
instead of Cmake.
Anybody forsee difficulties for compiling the whole acados with make?

roversch, 2017-04-11:
It's a lot of manual labor, CMake automates a lot of that.

I'm sure there are workarounds for those low-level chips, e.g. you could
specify a specific build type for them (CMAKE_BUILD_TYPE).

Could you give an example of what such a toolchain would look like?

zanellia, 2017-04-11:
Hey Dang, in my limited experience build systems can be a pain. CMake might help to manage builds as the project(s) grows in complexity. 

> Could you give an example of what such a toolchain would look like?
 
I would be also interested in knowing more about practical problems, if you are facing some already. I guess you could encounter problems whenever you have to use some high-level IDE or environment right? Otherwise you just need to have CMake installed on your host machine, configure the project and cross-compile, right?

doanminhdang, 2017-04-11:
This toolchain using Make is what I have worked with:
https://github.com/espressif/esp-idf/blob/master/make/project.mk

For each component, it specifies local additional make file.

> I prefer make, since for some low-level chips the toolchain supports Make
> instead of Cmake.
> Anybody forsee difficulties for compiling the whole acados with make?
>
>
>
> As @giaf <https://github.com/giaf> is not a big fun of cmake, I thought I
> could start an acados-based discussion here to convince him :P
>
> <https://github.com/acados/acados/issues/61#issuecomment-293361036>, or
> mute
> the thread
> <
VTmxt_YqY7tks5ru8hXgaJpZM4M6d7U
> >
>
>
>
> <https://github.com/acados/acados/issues/61#issuecomment-293363342>, or
mute
> the thread
MEx6NTZpJS1tNGsyrLks5ru8ojgaJpZM4M6d7U>

—
<https://github.com/acados/acados/issues/61#issuecomment-293364795>, or mute
the thread

zanellia, 2017-04-11:
I do see the point in the sense that integrating acados directly into the toolchain might be tricky. However, you could also cross-compile a static or shared library for acados and just link against it from your Makefile-based build system.

doanminhdang, 2017-04-11:
Hi Andrea,

Thanks for sharing your experiences, I also feel so.

My problem is that in order to fit with the code development framework of
the chip, I better use their settings for cross-compiling, which was
meticulously constructed with Make (https://github.com/espressif/esp-idf)
 . It will compile the components of the freeRtos, then my own code, then
pack them into one binary file for flashing.

Currently I still didn't use the advanced feature that the toolchain
provides, eg. Make with component-wise makefile. My current tactics is to
put all the source files into one folder and (let the toolchain) run a
dirty make for it.

doanminhdang, 2017-04-11:
Thanks for the suggestion, Andrea. I'll check if such linking could be done
easily with my toolchain.

giaf, 2017-04-12:
Yes I'm still a fan of make instead of cmake, nothing has changed in that regard ;) 

Make provides exactly what I want: a clean and low-level way to tell _exactly_ how the project build should be, and where I know _exactly_ what is going on (e.g. regarding dependencies). The flexibility provided by cmake is in my opinion a minus and not a plus, at least in the (low level) stuff I'm doing.

That said, I'm not against adding to the make an additional cmake if this can make stuff easier in ACADOS. But this cmake should mimic _exactly_ what the make does (e.g. fix the compiler to the ones that work, fix the compilation flags to the ones I know work well). But at this point I don't see any difference with respect to a make called using e.g.
make TARGET=X64_INTEL_HASWELL LA=HIGH_PERFORMANCE

@zanellia thanks for the effort of adding a cmake. 
As it is now, it only works when BLASFEO is used (when it is not, entirely different sources are employed).

Note that BLASFEO HP doesn't work with any compiler other than gcc and clang, since the assembly-coded kernels can be processed only by these compilers. 
HPMPC also doesn't work with e.g. visual studio (and I never tested with anything else than gcc and clang). In my opinion, removing the list of tested compilers to choose among is a kind of blind guess on what can happen with other compilers.

zanellia, 2017-04-12:
To be honest, @giaf might be totally right, although I still feel for
slightly larger projects (like acados) CMake might help to reduce
complexity. I agree that CMake is not very transparent sometimes, however
most of the time it does an OK job. What would you loose in terms of
control over the compiler's behavior with the simple CMake build system I
hacked together for example? You should be able to set compiler/linker
flags exactly the way you do it with a Makefile. Am I missing something?

As it is now, it only works when BLASFEO is used (when it is not, entirely
> different sources are employed).
>

Yes, sorry, it is not meant to replace the Makefile-based build system at
all at the moment :)

giaf, 2017-04-12:
@zanellia Maybe I can return the question to you :) What would such a CMake give more than the current makefile? What are the features that you would like to have?

zanellia, 2017-04-12:
> I had some troubles managing ExternalProjects based on Makefiles so I made a minimal port of hpmpc's build system to CMake.

This is the main reason I hacked together a CMake build system for hpmpc.

I personally find CMake easier and I would expect it to bring some advantages in the near future. Instead of comparing CMake and Makefile in terms of what functionalities they offer, I would focus on how easy it is to develop and maintain build systems based on the two approaches.

giaf, 2017-04-13:
What about discussing this once we are back to the office after Easter?

zanellia, 2017-04-13:
Yea, sure!

zanellia, 2017-04-21:
So, although I do like the idea of having all the external projects using CMake, as Gianluca pointed out, it might be difficult to keep Make- and CMake-based build systems in sync. This holds for hpmpc and blasfeo and it is even more of a relevant issue with other dependencies like qpOASES. Thoughts?

roversch, 2017-04-21:
Why would you want them to be in sync? With acados, we use the Makefile
generated by CMake, if you have hpmpc and/or blasfeo standalone you can
choose between this or the Makefile handwritten by @giaf.

zanellia, 2017-04-21:
Sure, but the CMake-based build system will have to be updated whenever the
Makefile-based does. @giaf has no interest in doing so and the same might
hold for any other dependency. We can definitely move everything to CMake,
but we should keep in mind that it will add a step in the integration
process of external software packages.

> So, although I do like the idea of having all the external projects using
> CMake, as Gianluca pointed out, it might be difficult to keep Make- and
> CMake-based build systems in sync. This holds for [...]

>
>
> <https://github.com/acados/acados/issues/61#issuecomment-296205555>, or
mute
> the thread
ABXrjEvBwCr4IdqW6fVyLbtYIvS3Lr8mks5ryLyngaJpZM4M6d7U>

—
<https://github.com/acados/acados/issues/61#issuecomment-296242266>, or mute
the thread

zanellia, 2017-06-17:
Hi all, since both @giaf and @roversch asked for CMake for hpmpc and blasfeo, you can find the link to my (very rough) version of the build system for hpmpc here:

[https://github.com/zanellia/hpmpc/blob/crosscompiling/CMakeLists.txt](url).

Analogously the one for blasfeo is here:

[https://github.com/zanellia/blasfeo/blob/crosscompiling/CMakeLists.txt](url).

They only work for a single target since I only needed that specific version when hacked CMake in. However, it should be very easy to extend to all the other targets.

giaf, 2017-06-17:
Yep I already found ;)

And I added cmake to HPIPM (that by the way now is only missing partial condensing, compared to HPMPC)

zanellia, 2017-06-17:
Awesome! Does it have partial tightening too? That was QUICK.

giaf, 2017-06-17:
nope there is no partial tightening, that will be your part of the fun ;)

giaf, 2017-06-17:
it has ipm for dense qp, ipm for ocp qp, (full) condensing from ocp qp to dense qp

giaf, 2017-06-18:
Now also BLASFEO has cmake @roversch 
@zanellia why the minimum version is 2.8.11?

giaf, 2017-06-18:
@roversch HPMPC won't get any cmake: as soon as HPIPM is up and running, we can remove HPMPC from ACADOS

zanellia, 2017-06-18:
@giaf it might be good to have stable version of hpmpc to be used in acados
before we move to hpipm, what do you think?

giaf, 2017-06-19:
the "stable version of HPMPC" is the one already used in ACADOS, and I'm not putting any more effort in improving it

and HPIPM is stable and ready to be used, you can give it a try ;)

giaf, 2017-06-19:
@zanellia how did you choose cmake minimum version 2.8.11 in your cmake (that I just copied for now, but I would like to know more about?)

zanellia, 2017-06-19:
I just copied that from acados? I do not remember to be honest.

giaf, 2017-06-19:
In my pc I have version 3.5. What is the oldest version you tested on?

Also, I had to replace all EQUAL with MATCHES in order to make it work properly

giaf, 2017-06-19:
I downloaded 2.8.0 and it fails with BLASFEO due to the assembly code.
Version 2.8.11 works, so I'll leave that as minimum version.

roversch, 2017-08-30:
Closing this in favour of #57

closed 2017-08-30 by roversch