acados/acados #73: Sometimes blasfeo build fails in CMake

roversch, 2017-04-21:
It's a bit weird, but very sporadically I get this in CMake on Travis, even if the other builds with other compilers all succeed:

```
The command "cmake -E chdir build cmake -D CMAKE_BUILD_TYPE=Debug .." exited with 0.
0.34s$ cmake --build build --clean-first
Scanning dependencies of target blasfeo_project
[  1%] Creating directories for 'blasfeo_project'
[  2%] No download step for 'blasfeo_project'
[  3%] No patch step for 'blasfeo_project'
[  4%] No update step for 'blasfeo_project'
[  6%] No configure step for 'blasfeo_project'
[  7%] Performing build step for 'blasfeo_project'
make[4]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
ar: ./auxiliary/d_aux_lib4.o: No such file or directory
make[3]: *** [static_library] Error 1
make[3]: *** Waiting for unfinished jobs....
make[4]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
make[4]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
make[4]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
make[4]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
make[2]: *** [external/blasfeo_project-prefix/src/blasfeo_project-stamp/blasfeo_project-build] Error 2
make[1]: *** [external/CMakeFiles/blasfeo_project.dir/all] Error 2
make: *** [all] Error 2
```

roversch, 2017-04-24:
@rienq also encountered this, with `gcc-5` on Travis. I cannot reproduce it on my machine.

roversch, 2017-05-08:
This occurred again today, see [this Travis build](https://travis-ci.org/acados/acados/jobs/230162095)

roversch, 2017-05-10:
@giaf this is fixed with [this](https://github.com/acados/acados/commit/d6ba36afdec79af7285900f9a164f1b0d8ac1e80) right?

zanellia, 2017-05-10:
I think so. Or at least it would fix the problem if the only issue was
parallelization of the build. Time will tell.

giaf, 2017-05-10:
yep in theory it should be fixed, I think we can close the issue, and open a new one if it happens again

closed 2017-05-11 by giaf