acados/acados #74: qpOASES zeroes an array when it's not necessary; wastes time

roversch, 2017-04-21:
I found something interesting while profiling `chen_nmpc_qpoases_example` on my Macbook Pro: almost 20% of the time is spent in zeroing an array (`_platform_bzero`).

[image]

I tracked it down to the following lines in `QProblemB.c:1335` (computeCholesky) in qpOASES:

	/* 1) Initialises R with all zeros. */
	for( i=0; i<NVMAX*NVMAX; ++i )
		_THIS->R[i] = 0.0;

Furthermore, these lines seem unnecessary. I can comment those lines out and all of the unit tests still work. Meanwhile, `chen_nmpc_qpoases_example` gets a speedup of roughly 20%. 

Could someone with a Linux machine confirm that they get the same speedup by commenting out those lines, e.g. @giaf @zanellia ? Make sure to pull the latest `master` and put the build type to `Release`.

roversch, 2017-04-21:
Next suspects are the many `d_free` and `d_zeros` calls in `ocp_qp_qpoases_condensing`. Rewrite after condensing interface is completed.

dkouzoup, 2017-04-24:
Hey Robin. I would be reluctant to change the qpoases code without a deeper understanding of the code. Skipping Initialization to zeros might work for some operating systems and/or problem instances but not for others. Perhaps you can ping Joachim about this. Of course all the mallocs from the acados side should be gone once the new interface is in place.

roversch, 2017-04-24:
I agree, it was not my intent to propose to change the qpOASES code. I was
just curious if it would give the same speedup with a Linux C library.

If that's the case, I will contact Joachim about this.

dkouzoup, 2017-04-24:
Alright! By the way, I suspect that malloc is much slower in mac than in linux. At least on my system. Ps: you can have a shot at Greg's server for the benchmarking.

roversch, 2017-05-10:
The connection with Greg's box is too slow to do that from here. Could someone just try this for me? It's basically 
 - commenting out that line, 
- building qpOASES and 
- running `chen_nmpc_qpoases_example` again and see if it changes anything (put number of repetitions 10000).

If it also occurs on Linux, I will then write to Joachim.

roversch, 2017-06-08:
I can confirm, roughly the same speedup (~25%) can be seen on a Linux box. E-mail to Joachim is underway.

closed 2017-06-08 by roversch