acados/acados #76: Installation requirements incomplete?

schoelst, 2017-05-05:
Hi,

when I tried to install acados on a freshly setup Ubuntu 17.04 with cmake installed I got the following error: 
```
$ cmake ..
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done

-- No build type chosen. Valid build types are:
   Release (default), Debug, Test, NoExternalLibs

 --- Configuring acados with build type: Release ---

-- acados will be installed in /home/tobi/local
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find SWIG (missing: SWIG_EXECUTABLE SWIG_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.7/Modules/FindSWIG.cmake:63 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  swig/CMakeLists.txt:22 (find_package)
```

-- Configuring incomplete, errors occurred!
See also "/home/tobi/Documents/acados/build/CMakeFiles/CMakeOutput.log".

Up to there everything seemed to have gone smoothly.

I tried installing swig which did not resolve the issue (so far). Now I get:

```
$ cmake ..             

-- No build type chosen. Valid build types are:
   Release (default), Debug, Test, NoExternalLibs

 --- Configuring acados with build type: Release ---

-- acados will be installed in /home/tobi/local
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.10") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.3", minimum required is "3") 
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3")
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.7/Modules/FindPythonLibs.cmake:255 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  swig/CMakeLists.txt:35 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/tobi/Documents/acados/build/CMakeFiles/CMakeOutput.log".
```

What am I missing? - Please let me know if you need further details.

Cheers,

Tobi

dkouzoup, 2017-05-05:
Do you have Python3 installed?

doanminhdang, 2017-05-05:
Hi Tobi,
Maybe you need python3-dev and python3-pip.
Dang

schoelst, 2017-05-06:
Hi,

Thank you for the quick answers and sorry for my delayed response!
Yep Python3 is and was installed.

Installing Python3-dev and Python3-pip did not solve the problem, though it got me further. After also installing Python3-numpy my understanding is, that it is now looking for casadi. I am guessing this refers to the Python-version, but I am not sure where it is looking for it..
Do you have a place where I can read this up?

Thanks,

Tobi

roversch, 2017-05-08:
Your best shot is looking at how [Travis](https://github.com/acados/acados/blob/master/.travis.yml) does it, especially the `install_packages.sh` script.

roversch, 2017-05-10:
I updated the [installation instructions](https://github.com/acados/acados#acados). @schoelst can you try again?

schoelst, 2017-05-10:
The installation succeeded now, Thank you!  The examples in `acados_root/build/bin/` seem to be working (produce meaningful output).
Though I had to set `SWIG_MATLAB=0` otherwise I get the following error:
```
$ -- No build type chosen. Valid build types are:
   Release (default), Debug, Test, NoExternalLibs

 --- Configuring acados with build type: Release ---

-- acados will be installed in /home/tobi/local
CMake Error at cmake/FindCasadiMatlab.cmake:29 (message):
  Casadi not found!
Call Stack (most recent call first):
  swig/CMakeLists.txt:41 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/tobi/Documents/acados/build/CMakeFiles/CMakeOutput.log".
```

I have MATLAB installed (sym-links in `/usr/local/bin exist`) and casadi is sitting in the MATLAB folder (i.e. in `/home/tobi/Documents/MATLAB/casadi`).

For the python example I had to add `/home/tobi/local/lib` to the `PYTHONPATH` and to install `scipy`, `matplotlib` through pip3 and `python3-tk` through apt. - Hope that helps.

Thank you all for your help!

zanellia, 2017-05-10:
Maybe you need the casadi folder to be in acados/external?

dkouzoup, 2017-05-10:
I think cmake detects where casadi is

roversch, 2017-05-10:
You need to add the path of the Matlab Casadi binaries to your $MATLABPATH,
similar
as for Python with $PYTHONPATH

schoelst, 2017-05-10:
So I created a MATLABPATH variable now (it did not exist before), and made it point to my casadi installation. That did not change the error though.
```
$ echo $MATLABPATH
/home/tobi/Documents/MATLAB/casadi
```

roversch, 2017-05-10:
Can you try running `source install_packages.sh` in the acados root folder? It should do everything automatically. If it still gives that error, please report back.

schoelst, 2017-05-11:
I just pulled the last version and re-ran the script:

```
$ source install_packages.sh
[sudo] password for tobi: 
gpg: keybox '/tmp/tmplf6cqe6j/pubring.gpg' created
gpg: /tmp/tmplf6cqe6j/trustdb.gpg: trustdb created
gpg: key 4F2D377DDCB1CB6E: public key "Launchpad PPA for GNU Octave" imported
gpg: Total number processed: 1
gpg:               imported: 1
OK
W: The repository 'http://ppa.launchpad.net/octave/stable/ubuntu zesty Release' does not have a Release file.
E: Failed to fetch http://ppa.launchpad.net/octave/stable/ubuntu/dists/zesty/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
Collecting numpy
  Using cached numpy-1.12.1-cp27-cp27mu-manylinux1_x86_64.whl
Collecting scipy
  Using cached scipy-0.19.0-cp27-cp27mu-manylinux1_x86_64.whl
Collecting matplotlib
  Downloading matplotlib-2.0.2-cp27-cp27mu-manylinux1_x86_64.whl (14.5MB)
    100% |████████████████████████████████| 14.5MB 60kB/s 
Collecting pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=1.5.6 (from matplotlib)
  Using cached pyparsing-2.2.0-py2.py3-none-any.whl
Collecting cycler>=0.10 (from matplotlib)
  Using cached cycler-0.10.0-py2.py3-none-any.whl
Requirement already satisfied: six>=1.10 in /usr/lib/python2.7/dist-packages (from matplotlib)
Requirement already satisfied: pytz in /usr/local/lib/python2.7/dist-packages (from matplotlib)
Requirement already satisfied: python-dateutil in /usr/local/lib/python2.7/dist-packages (from matplotlib)
Requirement already satisfied: functools32 in /usr/local/lib/python2.7/dist-packages (from matplotlib)
Collecting subprocess32 (from matplotlib)
Installing collected packages: numpy, scipy, pyparsing, cycler, subprocess32, matplotlib
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg/pip/req/req_set.py", line 784, in install
    **kwargs
  File "/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg/pip/req/req_install.py", line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg/pip/req/req_install.py", line 1064, in move_wheel_files
    isolated=self.isolated,
  File "/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg/pip/wheel.py", line 345, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg/pip/wheel.py", line 316, in clobber
    ensure_dir(destdir)
  File "/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg/pip/utils/__init__.py", line 83, in ensure_dir
    os.makedirs(path)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/numpy'
```

My understanding of this is: the OCTAVE repository does not have a version for Ubuntu 17.04 yet (which is what I am on). - No idea what went wrong with numpy there, I guess missing privileges.

Also I realized, that I could have been clearer before:
`source install_packages.sh` worked just fine, the only thing I had to do was to set  `SWIG_MATLAB=0`.

roversch, 2017-05-11:
thanks, @schoelst . A couple of things:

- octave is only needed for developers
- only Python3 is supported. I recommend to use a Python3 environment with `conda`; or use `pip3`. IIRC, you have all `numpy`, `scipy` etc. already
- more important is the Casadi Matlab part. Could you just copy paste that into your terminal?

roversch, 2017-05-16:
Any news @schoelst ? Could you successfully install acados?

schoelst, 2017-05-16:
Hi, sorry for not getting back to you about this!
I seems to have worked, but I was not able to test it yet. I'll do that tomorrow and let you know.
Thanks!

jdeschut, 2017-05-23:
Hi all, this has been a quite helpful post so far! I would like to ask about following related issues (installing acados for matlab and python on xubuntu).

- just as a report: when running cmake, I had to manually insert the path for MATLAB_ROOT in the /build/CMakeCache.txt file. After running cmake again, I additionally had to manually add the path to the different Matlab libraries to the MATLAB_LIBRARIES:PATH= ... line. Then things worked fine.

- during make install, I get following error from swig:
`
 [92%] Built target mass_spring_hpmpc_example
[ 93%] Swig source
swig error : Unrecognized option -matlab
Use 'swig -help' for available options.
swig/CMakeFiles/acadosMATLAB.dir/build.make:61: recipe for target 'swig/acadosMATLAB_wrap.cxx' failed
make[2]: *** [swig/acadosMATLAB_wrap.cxx] Error 1
CMakeFiles/Makefile2:584: recipe for target 'swig/CMakeFiles/acadosMATLAB.dir/all' failed
make[1]: *** [swig/CMakeFiles/acadosMATLAB.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`
When I type swig -help, the -matlab option is listed as a Target Language Option. 

Cheers,
Jochem

roversch, 2017-05-23:
Hi Jochem. You need to install swig from
https://github.com/jaeandersson/swig

Make sure to uninstall the one you already had.

rienq, 2017-05-23:
I also did not see this in the travis build script though, why is that?

roversch, 2017-05-23:
Look at [`install_packages.sh`](https://github.com/acados/acados/blob/master/install_packages.sh#L28)

rienq, 2017-05-23:
Yea, that's the script that I was talking about.. after running it and having installed things successfully, I still get the exact same error about the unrecognized -matlab option in swig
Then I saw your solution here above about installing swig from Joel's repo but this should then also have been done as part of this installation script or not? Or you first have to make sure that swig is uninstalled maybe before running the install_packages script?

roversch, 2017-05-23:
Sorry, I was unclear. You should install it from Joels' repo, but it is included as a git submodule, so you should have it already. Then, follow the instructions as in `install_packages.sh`.

Make sure that you have the `matlab` branch of SWIG!

Also, did you add `acados/external/swig` to the front of your `PATH` environment variable?

jdeschut, 2017-05-24:
Solved. Thanks Robin!

rienq, 2017-05-24:
Hmm, did you remove all other swig installations to make this work? If I do that, CMake complains it cannot find swig.. if I don't remove swig3.0, then I get the same error about the -matlab option even though I do have Joel's version of swig as an acados submodule and I do have the PATH set correctly as well

roversch, 2017-05-25:
Maybe delete your `build` folder and do a clean build? I guess you already did that.

Did you do an in-source or out-of-source build of `swig`?

schlagenhauf, 2017-05-31:
Hi guys,
i also had troubles installing acados, i made an attempt to make the installation instructions more verbose, in addition to extending the install script to perform the whole installation, see #83. Hope that helps.

rienq, 2017-06-02:
Great, thank you all! The new installation script works out of the box on Ubuntu 16.04.. ;-)
However, for calling the matlab examples, I guess you also need to add the swig generated files and the acadosMEX file to the MATLAB path right? So this could also be added to the installation script and the README file in that case..

roversch, 2017-06-02:
You're right, I added a few comments on that. 

By the way, the `MATLABPATH` environment variable is only read when you start `matlab` from a terminal, as far as I know.

dkouzoup, 2017-06-03:
Can we not add those paths in the beginning of each python/matlab script instead of using env. variables?

roversch, 2017-06-04:
But what if a user has a different installation directory than `~/local/lib`? Then the scripts won't work..