acados/acados #80: Unifying memory and workspace usage among QP solvers

zanellia, 2017-05-11:
We just had a short discussion about memory and workspace usage in QP solvers together with @giaf  and @dkouzoup. In particular, @giaf pointed out that, by storing stuff like cost and dynamics matrices in the workspace, the overall memory footprint could be reduced (given that the memory associated with the QP workspace could be reused for other modules (including other QP solvers), while memory allocated for the QP memory could not). On the other side, this complicates things from a user perspective and could cause inefficiency in some cases when cost  and dynamics matrices should be fixed (e.g. in linear MPC or inexact methods where part of the sensitivities are fixed). We eventually agreed on storing this kind of things in the memory to simplify things, but we also thought some other people might have a different opinion and that it could be fruitful to discuss this now as we start unifying memory and workspace usage. Any comment/advice is more than welcome!

roversch, 2017-05-11:
It's good to keep this in the back of our heads if we ever run into memory problems (e.g. on some embedded systems), but I would not prematurely optimize for memory footprint. How big of a gain in memory usage are we talking here? How much can be reused by other modules?

dkouzoup, 2017-05-16:
It's hard to say "how-much" since it is also solver-dependent. I agree though that we should first aim for efficiency and then for memory footprint. Plus supporting both would bloat the code. So if there are no further objections, we can store all data that might be reused in the memory and close the issue.

roversch, 2017-08-30:
Closing for now, we'll go for the way that uses more memory but is more convenient.

closed 2017-08-30 by roversch