acados/acados #82: invalid mex-file in matlab interface

jdeschut, 2017-05-24:
Hi all, after successful installation of acados for matlab and python (python examples run without problems), I encountered following error when running ocp_nlp_example.m in the matlab interface:

```
Invalid MEX-file '/home/jochem/local/lib/acadosMEX.mexa64':
Missing symbol 'dlerror' required by '/home/jochem/local/lib/acadosMEX.mexa64'
Missing symbol 'dlopen' required by '/home/jochem/local/lib/acadosMEX.mexa64'
Missing symbol 'dlsym' required by '/home/jochem/local/lib/acadosMEX.mexa64'
Missing symbol '_ZNSt13runtime_errorC1EPKc' required by '/home/jochem/local/lib/acadosMEX.mexa64'
Missing symbol '_ZNSt16invalid_argumentC1EPKc' required by '/home/jochem/local/lib/acadosMEX.mexa64'.

Error in acados.ocp_nlp (line 229)
        tmp = acadosMEX(106, varargin{:});

Error in ocp_nlp_example (line 8)
nlp = ocp_nlp(struct('N', N, 'nx', nx, 'nu', nu));
```

Any idea if this could still be installation-related?
 
Cheers,
Jochem

roversch, 2017-05-25:
Which compiler do you have? Is it supported by MATLAB?

 https://www.mathworks.com/support/compilers.html
<https://www.google.com/url?q=https%3A%2F%2Fwww.mathworks.com%2Fsupport%2Fcompilers.html&sa=D&sntz=1&usg=AFQjCNGNR3QbnczM5LjDzSG9jmI8Zz5lxg>

jdeschut, 2017-05-26:
This solved the problem: MATLAB only supports up to gcc-4.9. After re-installing with the right compiler (using `update-alternatives`), the test example runs. Thanks!

closed 2017-05-26 by roversch