acados/acados #84: Unit testing for HPMPC interface

dkouzoup, 2017-06-02:
Enabling the TEST_HPMPC flag in test_qpsolvers.cpp and running the unit tests causes a segfault. Moreover, although the call of the QP solver seems to follow our newest convention the free_memory call is missing, which may cause memory leaks (or hmpc has no memory at all atm).

dkouzoup, 2017-06-02:
A happy birthday issue for @zanellia ;) Brought to you by your dear colleague in cooperation with @rienq.

rienq, 2017-06-02:
Yes, Happy birthday from me too! Also, get this fixed asap.. thank you! :D

zanellia, 2017-06-06:
Thanks all for the lovely birthday gift! I just came back from holidays and I am working on it ;) Is there any chance the error was not a segfault, but a floating point exception? This is what I get running the unit test on my laptop:

> /home/andrea/acados/test/ocp_qp/test_qpsolvers.cpp:187: FAILED:
  {Unknown expression after the reported line}
due to a fatal error condition:
  SIGFPE - Floating point error signal

I generally get SIGFPEs when running hpmpc with floating point exceptions on, but never looked into a solution before. Can @dkouzoup confirm that he also experiences the same problem?

dkouzoup, 2017-06-06:
Hey Andrea. I am sorry, I was too quick. Indeed it's a SIGFPE problem (at least on my linux machine, I will check on mac too). Is there a workaround to fix this or do we just re-assign the issue to @giaf? ;)

zanellia, 2017-06-06:
Seems like there is an arithmetic exception in d_part_cond_compute_problem_size_libstr. I am trying to fix it ;)

zanellia, 2017-06-06:
Alright, two things: 1) N2 was not set by default in gian's interface, so there was a divide-by-zero in the computation of workspace size 2) hpmpc's result are accurate up to 10e-5, so I had to lower the accuracy. Waiting for Travis to test the build, will merge in few minutes.

zanellia, 2017-06-06:
https://github.com/acados/acados/commit/a8ba716d707cd317093908f4c3dd1c43c65f7a6 fixes unit testing for hpmpc.

closed 2017-06-06 by zanellia