acados/acados #90: Support for dSpace

roversch, 2017-06-09:
Once we have the Simulink interface #89 and Windows support #88 , we can start thinking about how to bring this all together for dSpace and try it out on one of their platforms.

zanellia, 2017-06-21:
@rienq what is the current state with #88 ? Together with @doanminhdang we might have found a nice and clean way to port acados to dSPACE. Instead of dealing with the Real-Time Interface directly we can just cross-compile everything using the default build system and link against a shared library when deploying to the dSPACE box. That would be much nicer then the setup we had in ACADO, where a totally different (and rather messy) build system had to be used to compile for dSPACE.

mvukov, 2017-06-21:
> That would be much nicer then the setup we had in ACADO, where a totally different (and rather messy) build system had to be used to compile for dSPACE

Ha, ha :)

Anyway, can you link shared libs to dSpace binaries?

zanellia, 2017-06-21:
Sorry Milan do not take it personally :) The build system for ACADO is was it is because of how the Real-Time Interface work. 

> Anyway, can you link shared libs to dSpace binaries?

Apparently yes. That is what they do in FORCES. It is still to be assessed whether the new workflow is as painless as I expect it to be ;)

rienq, 2017-06-21:
Sounds great! The current status of the Windows support is as @roversch described in his recent e-mail on the mailing list: we can create libraries that work on Windows based on MinGW.. @roversch also fixed the CMake scripts such that we could in principle support MSVC but we would have to move back to C89 instead of C99 which we might not want to do if we can avoid it.. ;-)

roversch, 2017-06-21:
See #96 , which I just accepted..

mvukov, 2017-06-21:
> Sorry Milan do not take it personally :) 

I didn't :) The build system was always kinda complicated, PhD-grade level :)

> That is what they do in FORCES.

Interesting. Are you referring to .o or .so file that FORCES produces?

zanellia, 2017-06-22:
> Anyway, can you link shared libs to dSpace binaries?

Actually I do not know. I should have better specified: for the dSPACE MicroAutoBox it might not be possible. However, the point here is to be able to generate binaries (not necessarily shared libs) using the default build system. 

Moreover, the workflow for the MicroAutoBox uses a quite obsolete compiler (Mentor Graphics mccppc). Standalone cross-compilation might not be the best choice for this platform. Other dSPACE platforms use instead a gcc-based cross-compiler for which the workflow I propose might be a good option.

rienq, 2017-06-22:
Wait, so what is it then that FORCES already does in this way?

And I agree with the obsolete compiler for MABX but I am afraid there is not much we can do about that, we would even potentially "lose" a lot of users if our code could not run on such embedded platforms

zanellia, 2017-06-22:
embotech has set up cross-compilation on their servers such that you only get an object file (instead of a shared library) to link against. The problem is that cross-compiling 2 or 3 files like we had in ACADO is one thing, cross-compiling a medium size cmake-based project is another. I am trying right now to see how far I can get with this approach, but I am not sure is the way to go for MABX. @rienq do you have any experience to share here? 

The alternative is to setup a completely independent build system like Milan did with the Legacy Code Tool or a hand-written .mk file. The potential problem, again, is that you need to maintain both build systems and that their complexity  is not anymore the one of the generated code in ACADO. We are not talking about a huge project (yet), but it is not going to be a 10 minutes thing anymore :)

For other platforms, where cross-compilation is much more straightforward,  it should be much easier and we could in principle use the very same build system.

In general I think portability of the build system is something we did not take into account that much so far. It might be good to discuss this a bit more into details.

roversch, 2017-06-22:
> In general I think portability of the build system is something we did not take into account that much so far. It might be good to discuss this a bit more into details.

This might be a good idea indeed. Before we do that, I want to understand what exactly the issue is. CMake does support other compilers than gcc, so couldn't you just hack some CMake together to cross-compile binaries for dSpace? I have the feeling I'm missing the point :)

zanellia, 2017-06-23:
CMake does support other compilers, but I doubt it would be easy to make it work with a compiler that gives <600 results in a Google query ("mccppc").  And I would be surprised if it worked out of the box without being supported explicitly: the flags for this compiler are completely different than what gcc-like compilers do. -l does not mean "link against this library", -O0/1/2/3 does not mean "optimize for a certain performance index" and so on. Of course we could hack something together, but it does not sound like it is the best option to me. 

To make it clearer, if we were to use a DS1006 dSPACE box, which uses a gcc-based compiler, we could just specify a different compiler in CMake and that's it. But it is not the case for the MicroAutoBox.  

In conclusion, we are working on a custom solution for this platform. For any platform using a gcc-like compiler (or cross-compiler) CMake will do the job ;)

doanminhdang, 2017-06-23:
Using dSpace's dsmake build system, I can compile the library for a small example of acados.

It may not be straightforward to let CMake interact with the dsmake (which is the old OPUS Make: http://opussoftware.com/tutorial/tutorial.htm), however our support for dSpace MicroAutoBox could be like this:
- we provide a specific additional make file, inside that file there is the listing of all source files of acados
- the user need to supply that additional make file as "user's make file" in the dSpace compilation system
- we also provide some S-functions so the user put in Simulink

The first two tasks are doable now, the S-function is waiting for @zanellia .

rienq, 2017-06-23:
Sounds great, Dang!
And what about the e-mail from @giaf who apparently managed to build BLASFEO/HPIPM using MinGW-w64 but still link it together with other source code in Visual Studio, could this workflow potentially not work also for the Microtec PowerPC C Compiler?

giaf, 2017-06-23:
Wait, I did something different!
MinGW-w64 was only used to assembly the .S assembly files into .o files.
Then all the .c files and the libraries were build in Visual Studio!

In my case, the only issue was that Visual Studio could not process the .S assembly files.

rienq, 2017-06-23:
I see, but if Visual Studio "understands" the object files from MinGW.. should we not be able to push this further and compile all of acados using our Cmake build system and later link the resulting files with an S-function?

doanminhdang, 2017-06-23:
I didn't try the MinGW neither Visual Studio, Rien. From tomorrow till the end of next week I'm traveling and will not touch that yet.

There are two schemes to compile with this Microtec PowerPC C compiler: 
1. We can compile the acados separately (as I'm doing now) and then provide a library, which then the S-function will refer to compiled functions in acados library, then it's dSpace's job to link everything when it compiles the RTI target. (I think FORCES is doing this way)
2. We provide the source code, so that the user will include it to compile together with dSpace's make for the RTI target (the procedure I suggested above).

Doing the 2nd way, we need to rely on dSpace build system, the only thing we could do is to give lists of source files, list of folders, and some compiler flags.

Doing the 1st way, @zanellia thinks it is not good, because we can only use dynamic memory allocation. If some algorithms require static memory allocation then it is not possible to provide pre-compiled lib to the users.

giaf, 2017-06-23:
I just tested that Visual Studio understands the object files from my assembly...
Most likely this doesn't work if in the .c files there are system calls and maybe even structures (that may have different padding).

zanellia, 2017-06-23:
We are talking about a totally different compiler

rienq, 2017-06-23:
I thought the whole point was to find a way similar to the first way, such that we can reuse the cmake build system instead of making a second build system that we need to maintain (which is the case for the second option).. ? option 2 is always possible and is pretty much the "ACADO dirty way" if you want ;-)

I don't think static memory is a problem for the first approach, is it? Since you could generate a header file with problem dimensions and let the S-function point to static memory depending on those problem dimensions that can be defined as compiler preprocessor directives (similar to ACADO generated code), I have already done similar things here at MERL using acados code in order to hack some interfaces together and compare some solvers

rienq, 2017-06-23:
Yes, but if it does not work with Visual Studio then for sure it won't work with Microtec PowerPC so we are just trying to understand what are the limitations of both of them.. ;-)

rienq, 2017-06-23:
Remember that we have to rewrite all the code to comply with C89 instead of C99 if we do decide to build acados source code with Microtec PowerPC to support dSPACE

zanellia, 2017-06-23:
@rienq yes that was the point, but I do not think it is possible for the
MicroAutoBox. And I think Dang refers to the possibility of providing a
precompiled static library such that the user does not have to bother with
the build system. In that case you do need dynamic allocation since the
problem size is not known.

giaf, 2017-06-23:
The way ACADOS was initially thought to be built was to avoid _any_ memory allocation in the library exactly to avoid this kind of issues. All memory allocation would just take place in the main file outside the library...

zanellia, 2017-06-23:
I think the C89/C99 is yet another issue. The problem here is the fact that
the only build system we know of (and probably the only one) that supports
the Microtec compiler is very limited (basically the one that ACADO was
using).

rienq, 2017-06-23:
Exactly, @giaf, and the main file would now be an S-function which instead of using malloc, it just points to static memory.. I still don't see the problem with this? The user would not have to bother with our build system, only with building the S-function in that case, right?

(Whether it is possible, I am indeed not sure either.. but clearly it would be a more ideal situation)

zanellia, 2017-06-23:
@giaf well, if you need to do dynamic memory allocation in the main file
you can as well do it using acados's functions...which is by the way how we
do it now. In that way you can avoid recompiling acados everytime you
change dimensions for example

rienq, 2017-06-23:
That's still unrelated, I use acados all the time with changing dimensions and I never rebuild acados for that.. why would you?

zanellia, 2017-06-23:
Originally, we were thinking to have fixed dimensions specified in a header
such that everything is known at compile time. But I think the discussion
is deviating from the main point :)

rienq, 2017-06-23:
That is what I am saying as well, but you do not need to recompile acados for that because acados does not take care of the memory, your main-file or S-function does, so simply change those header files (however that is, manual or automated), recompile your S-function and link to acados libraries.. why would you recompile acados?
(and yes, we are totally deviating :D)

doanminhdang, 2017-06-23:
If there is no need to do memory allocation in the acados lib, then we go for the first way, is there any more issues with this? I just don't understand cmake yet to know how to generate (at the end) command lines for the Microtec C compiler to compile each C source.

About C89 and C99 code, I also wish to get C89 code. At this point, when I tried porting thr example mass_spring_hpmpc yo MicroAutoBox, the compiler does not complain.

roversch, 2017-06-23:
> the flags for this compiler are completely different than what gcc-like compilers do. -l does not mean "link against this library", -O0/1/2/3 does not mean "optimize for a certain performance index" and so on.

@zanellia : Our CMake should _not_ depend anywhere on those flags, you should rely on `target_include_directories` and `CMAKE_BUILD_TYPE` instead.

Then again, you are right that the `mccppc` compiler probably does not know how to handle these. But CMake is built to be extensible, so we should be able to write a [Toolchain file](https://cmake.org/Wiki/CMake_Cross_Compiling) specifically for that purpose (with `CMAKE_SYSTEM_NAME=Generic`). I did something similar for `mingw-w64`, see https://github.com/acados/acados/blob/master/cmake/Toolchain-mingw32.cmake 

That is, when we decide to go the CMake route, which is not at all clear atm :)

roversch, 2017-06-23:
On a related note: I'm not at all convinced that this `mccppc` is not able to compile C99 code and it seems from Dang's experiment that it actually does. I phoned dSpace support a while ago and they told me it was possible "in principle". But let's wait and see if it actually does :)

zanellia, 2017-06-23:
As far as I know a basic tool chain file only specifies the location of
compiler, linker and search paths.

CMake needs to generate instructions for compiler and linker at some point
relying on some conventions. What if the conventions for mccppc are not
supported(which does not seem to be the case)? Am I missing something?

And sure, CMake is extensible, the question is how hard it would be to
extend it and if it is worth it.

C89/C99 is yet another thing. Actually dang was able to compile, so I would
say it might work.

zanellia, 2017-06-23:
What I am trying to say is that we might need to do is basically to add
support for a new "language" in CMake:

https://cmake.org/pipermail/cmake/2011-March/043417.html

Now the question is question is just whether we want to do this or have a
separate build system for MABX.

roversch, 2017-06-23:
@zanellia You're absolutely right, and it is probably not worth it to implement a new language.

How do the compile commands look like for `mccppc` anyway?

Also, FYI: http://s3.mentor.com/public_documents/whitepaper/resources/mentorpaper_49867.pdf , so C99 style comments are accepted, just like in-for variable initialization (`for(int i = 0; ...)`)

roversch, 2017-06-23:
Actually, from this [probably deprecated manual](ftp://ftp.software.ibm.com/software/rational/docs/documentation/manuals/OTD/portingguidehtml/appendia.jsp#10633), it seems `mccppc` is similar to gcc, but with some annoying changes (-J include tag instead of -I).

zanellia, 2017-06-23:
> Actually, from this probably deprecated manual, it seems mccppc is

Exactly. I guess with a good knowledge of what CMake does under the woods a
few changes might suffice actually. Let's say we give it a shot on Monday
and if it turns out to be a complex task we hack something together without
CMake?

roversch, 2017-06-23:
> Let's say we give it a shot on Monday
and if it turns out to be a complex task we hack something together without
CMake?

Agreed. Do you have a compiler specification? Like what options etc?

roversch, 2017-06-23:
After hacking around in CMake a bit, I'm more convinced that what you suggest would be feasible. For example,

    set(CMAKE_INCLUDE_FLAG_C "-J")

gives me 

    /usr/bin/cc  -J/Users/robin/local/include  -std=gnu90 etc. etc.

generated by CMake with the "Unix Makefiles" generator. It's cleaner to write our own generator for `dsmake`, but hey who has time for that? :p

roversch, 2017-08-11:
I was able to 'hack' the dSpace tools into CMake, see this pull request for those that are interested: #109

FreyJo, 2020-01-16:
Closing, recent S function workflow using templates can be found here:
https://github.com/acados/acados/pull/532

closed 2020-01-16 by FreyJo