acados/acados #99: Request to relax lint

giaf, 2017-07-01:
The under-signer giaf thereby officially makes a request to relax the lint requirements, limiting them to the strictly necessary to keep readability of the code (e.g. spaces instead of tabs, max 100 characters, curly brackets alignment) and e.g. letting every coder free to choose how many spaces/white lines (s)he may feel the need to put in the code (s)he is working on, without having to waste more time on these formal coding requirements than in the coding itself.

Yours faithfully,

giaf

dkouzoup, 2017-07-03:
I feel sympathy for the concerns of the above-signer and given his current and upcoming contributions we should at least discuss it in an upcoming meeting ;) What I find annoying myself are the warnings to errors on unused variables, since they create a lot of dummy code all over the place.. Is it in principle possible to prevent only those warnings from becoming errors?

roversch, 2017-07-13:
I'm a bit reluctant to relax the conditions, could you first try to follow [these steps ](https://www.quora.com/How-can-I-configure-Vim-to-use-the-Google-C++-coding-style-guide)to make your workflow easier?

roversch, 2017-07-24:
You convinced me to remove the '-whitespace/blank_line' rule. #fe45789ec027bc6c19c045fba120362d5a4f35f7

closed 2017-08-10 by roversch