acados/acados #104: acados install: allow using pre-installed Casadi

doanminhdang, 2017-08-04:
Current build system of acados needs to search for Casadi binary, downloaded from Casadi website. The binary provided by Casadi is limited to Intel CPU, not compatible with ARM processors.
I just compiled acados for Raspberry Pi 3 running Raspbian. Since Casadi binary is not compatible, I need to compile Casadi and install it to the system. To compile acados, I changed the file FindCasadiPython.cmake to set Casadi directories manually:

> find_package(PythonInterp 3 REQUIRED)
> 
> # Try to find Casadi root directory
> #find_path(
> #    CASADIPYTHON_ROOT_DIR
> #    NAMES "casadi/casadi.py"
> #    PATHS ENV PYTHONPATH
> #)
> #if (NOT CASADIPYTHON_ROOT_DIR)
> #    message(FATAL_ERROR "Casadi not found!")
> #endif ()
> 
> # Determine the version number
> execute_process(
>     COMMAND "${PYTHON_EXECUTABLE}" -c "from sys import path; path.append(r'${CASADIPYTHON_ROOT_DIR}');import casadi; print(casadi.CasadiMeta_getVersion())"
>     OUTPUT_VARIABLE CASADIPYTHON_VERSION
> )
> string(STRIP "${CASADIPYTHON_VERSION}" CASADIPYTHON_VERSION)
> string(SUBSTRING "${CASADIPYTHON_VERSION}" 0 1 CASADIPYTHON_MAJOR_VERSION)
> string(COMPARE EQUAL "${CASADIPYTHON_MAJOR_VERSION}" "3" FOUND_CASADIPYTHON_3)
> if (NOT FOUND_CASADIPYTHON_3)
>     message(FATAL_ERROR "Casadi version 3 required. Found version: ${CASADIPYTHON_VERSION}")
> endif ()
> 
> # Find Casadi libraries
> #set(CASADIPYTHON_LIBRARY "${CASADIPYTHON_ROOT_DIR}/casadi")
> set(CASADIPYTHON_LIBRARY "/usr/local/lib")
> #find_path(
> #    CASADIPYTHON_INCLUDE_DIR 
> #    NAMES casadi/casadi.hpp
> #    PATHS ${CASADIPYTHON_ROOT_DIR}/include
> #)
> set(CASADIPYTHON_INCLUDE_DIR "/home/casadi/build/compiled/include")
> include(FindPackageHandleStandardArgs)
> find_package_handle_standard_args(CasadiPython DEFAULT_MSG CASADIPYTHON_LIBRARY CASADIPYTHON_INCLUDE_DIR)

rienq, 2017-08-04:
It sounds great that you managed to install acados on RPI this way, but I am not sure whether it really makes sense to use acados with CasADi on an ARM processor or any other "more embedded" platform in the first place.. this would only make sense if you would want to use acados from Matlab/Python right? While it makes probably more sense to use acados from C in that case.. or am I missing something here?

doanminhdang, 2017-08-04:
Indeed I don't intend to run Casadi on RPI. It's because to have the Python
interface for acados, the compiler process also want me to have Casadi.
Yesterday I installed all, but only need to run Python examples of acados.
(I don't know whether Casadi is run behind in those ocp_qp.py and
ocp_nlp.py)

The Python interface is handy. That will help me (and other RPI users)
evaluate acados. If within a day I can run acados and interface with
hardware in RPI (most of tutorials on the web for RPI use Python to
interact with hardware), then it'll be attractive.

<https://github.com/acados/acados/issues/104#issuecomment-320296895>, or mute
the thread

roversch, 2017-09-13:
This can be circumvented, without hacking the FindCASADIPYTHON.cmake file, by setting the `PYTHONPATH` environment variable to point to your Casadi installation.

Can you try that please?

closed 2017-09-13 by roversch