acados/acados #110: ImportError: cannot import name 'ocp_nlp_solver'

Flancis, 2017-08-29:
After installing the tool as instructed in the manual, I tried to run an example but it keeps giving the following error code:
Traceback (most recent call last):
  File ".../examples/python/ocp_nlp_new.py", line 5, in <module>
    from acados import ocp_nlp, ocp_nlp_solver
ImportError: cannot import name 'ocp_nlp_solver'

I've already set up  the python environmental variables. Anyone has any idea what happens here?

Thanks.

roversch, 2017-08-29:
Can you give some more info? What OS+version?

Flancis, 2017-08-29:
Ubuntu 16.04. It runs on VirtualBox.

Flancis, 2017-08-29:
And I'm using it in python.

roversch, 2017-08-29:
Can you check the file `acados.py`? Is the class `ocp_nlp_solver` defined in there?

Flancis, 2017-08-29:
Could you tell me where is that file? I didn't find it in the package.

roversch, 2017-08-29:
Sorry, I should have been more clear :)

It should be in your acados installation (by default `~/local/lib/`)

Flancis, 2017-08-29:
I installed acados into another folder, and in that folder I didn't find the file you mentioned. Also, in the folder `ocp_nlp`, there is no file named `ocp_nlp_solver`, which is very strange. The only file named `acados` I could find is `acados.i`, and in that file, the  `ocp_nlp_solver` is not defined.

By the way, I used the command `./install.sh` to install this package.

roversch, 2017-08-29:
But it does seem to find `ocp_nlp`, right? Could you double check this by typing `from acados import ocp_nlp`?

Flancis, 2017-08-29:
Yes. When I run `from acados import ocp_nlp` in python3 idle, no error code comes out. What then?

roversch, 2017-08-29:
Do you have Python3.5+?

Flancis, 2017-08-29:
Yes. Python 3.5.2 is installed as suggested.

roversch, 2017-08-29:
Weird. I'm trying to recreate the error now and will get back to you with more info.

roversch, 2017-08-29:
Hi @Flancis , I just retraced your steps and installed `acados` in a clean Ubuntu 16.04 Virtual Machine and I get no problems whatsoever. To debug this a bit further:

Could you attach your `install_manifest.txt`? That should be in your build folder. 

Then, could you try a clean install, by removing the build folder and running the following from the acados root folder?

    mkdir build
    cd build
    cmake -D SWIG_PYTHON=ON ..
    make
    ctest

Flancis, 2017-08-29:
It's very strange that I didn't find that file either. I tried what you said, and it gave the following error:

`-- MATLAB not found. Set MATLAB_ROOT
-- Will look for MATLAB libraries in 
-- MATLAB not found. If you do have it, set MATLAB_ROOT and reconfigure (missing:  MATLAB_ROOT MATLAB_INCLUDE_DIR MATLAB_LIBRARIES MATLAB_MEX_PATH MATLAB_MEXEXT_PATH MATLAB_MEX_EXT) 
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3")
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindPythonLibs.cmake:265 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  swig/CMakeLists.txt:87 (find_package)

-- Configuring incomplete, errors occurred!`

It says that `missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS`. I have python 3.5.2 installed and tried the method on this forum but it still didn't work.

roversch, 2017-08-29:
So probably the install failed before as well?

Try the following please:

    sudo apt-get install python3-dev
    cmake -D SWIG_PYTHON=ON -D PYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1 -D PYTHON_INCLUDE_DIR=/usr/include/python3.5m/ .. 
    make
    ctest

> those paths worked for me on my VirtualBox, you might have to alter them.. Basically, the problem is that this version of `CMake` does not recognize more modern versions of Python.

Flancis, 2017-08-29:
Yeah, it probably failed in the first place. Anyway, it now worked until the last step. When running ctest, the following error codes come out.
The following tests FAILED:
	  5 - ocp_nlp_py (Failed)
	  6 - ocp_qp_py (Failed)
	  7 - sim_py (Failed)

roversch, 2017-08-29:
Okay, we are getting closer. could you try 

`ctest --output-on-failure`

and paste the output here please?

Flancis, 2017-08-29:
Yeah, the output is as follows.
Test project /home/zw/acados0828/build
    Start 1: mass_spring_hpmpc_example
1/7 Test #1: mass_spring_hpmpc_example .............   Passed    1.28 sec
    Start 2: mass_spring_qpoases_example
2/7 Test #2: mass_spring_qpoases_example ...........   Passed    0.09 sec
    Start 3: chen_nmpc_qpoases_example
3/7 Test #3: chen_nmpc_qpoases_example .............   Passed    1.51 sec
    Start 4: pendulum_partial_tightening_example
4/7 Test #4: pendulum_partial_tightening_example ...   Passed    0.01 sec
    Start 5: ocp_nlp_py
5/7 Test #5: ocp_nlp_py ............................***Failed    0.39 sec
Traceback (most recent call last):
  File "/home/zw/acados0828/build/swig/acados.py", line 18, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 906, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: /home/zw/acados0828/build/swig/_acados.so: undefined symbol: _ZN6casadi13GenericMatrixINS_6MatrixINS_6SXElemEEEE3symERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8SparsityE

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zw/acados0828/examples/python/ocp_nlp.py", line 5, in <module>
    from acados import ocp_nlp, ocp_nlp_solver
  File "/home/zw/acados0828/build/swig/acados.py", line 21, in <module>
    _acados = swig_import_helper()
  File "/home/zw/acados0828/build/swig/acados.py", line 20, in swig_import_helper
    return importlib.import_module('_acados')
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/zw/acados0828/build/swig/_acados.so: undefined symbol: _ZN6casadi13GenericMatrixINS_6MatrixINS_6SXElemEEEE3symERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8SparsityE

    Start 6: ocp_qp_py
6/7 Test #6: ocp_qp_py .............................***Failed    0.19 sec
Traceback (most recent call last):
  File "/home/zw/acados0828/build/swig/acados.py", line 18, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 906, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: /home/zw/acados0828/build/swig/_acados.so: undefined symbol: _ZN6casadi13GenericMatrixINS_6MatrixINS_6SXElemEEEE3symERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8SparsityE

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zw/acados0828/examples/python/ocp_qp.py", line 3, in <module>
    from acados import ocp_qp, ocp_qp_solver
  File "/home/zw/acados0828/build/swig/acados.py", line 21, in <module>
    _acados = swig_import_helper()
  File "/home/zw/acados0828/build/swig/acados.py", line 20, in swig_import_helper
    return importlib.import_module('_acados')
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/zw/acados0828/build/swig/_acados.so: undefined symbol: _ZN6casadi13GenericMatrixINS_6MatrixINS_6SXElemEEEE3symERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8SparsityE

    Start 7: sim_py
7/7 Test #7: sim_py ................................***Failed    0.21 sec
[... 11 lines cut ...]
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: /home/zw/acados0828/build/swig/_acados.so: undefined symbol: _ZN6casadi13GenericMatrixINS_6MatrixINS_6SXElemEEEE3symERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8SparsityE

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zw/acados0828/examples/python/sim.py", line 3, in <module>
    from acados import sim_solver
  File "/home/zw/acados0828/build/swig/acados.py", line 21, in <module>
    _acados = swig_import_helper()
  File "/home/zw/acados0828/build/swig/acados.py", line 20, in swig_import_helper
    return importlib.import_module('_acados')
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/zw/acados0828/build/swig/_acados.so: undefined symbol: _ZN6casadi13GenericMatrixINS_6MatrixINS_6SXElemEEEE3symERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8SparsityE

57% tests passed, 3 tests failed out of 7

Total Test time (real) =   3.68 sec

The following tests FAILED:
	  5 - ocp_nlp_py (Failed)
	  6 - ocp_qp_py (Failed)
	  7 - sim_py (Failed)
Errors while running CTest

roversch, 2017-08-29:
This is a known bug in gcc-5. I pushed a fix. Could you

`git clone` and `make install` again?

Flancis, 2017-08-29:
Now two of them failed.
Test project /home/zw/acados0829/build
Cannot create directory /home/zw/acados0829/build/Testing/Temporary
Cannot create log file: LastTest.log
    Start 1: mass_spring_hpmpc_example
1/7 Test #1: mass_spring_hpmpc_example .............   Passed    1.30 sec
    Start 2: mass_spring_qpoases_example
2/7 Test #2: mass_spring_qpoases_example ...........   Passed    0.09 sec
    Start 3: chen_nmpc_qpoases_example
3/7 Test #3: chen_nmpc_qpoases_example .............   Passed    1.52 sec
    Start 4: pendulum_partial_tightening_example
4/7 Test #4: pendulum_partial_tightening_example ...   Passed    0.01 sec
    Start 5: ocp_nlp_py
5/7 Test #5: ocp_nlp_py ............................***Failed    0.42 sec
cc: error: vde.c: No such file or directory
cc: fatal error: no input files
compilation terminated.
Traceback (most recent call last):
  File "/home/zw/acados0829/examples/python/ocp_nlp.py", line 14, in <module>
    nlp.set_model(ode_fun, step)
  File "/home/zw/acados0829/build/swig/acados.py", line 446, in set_model
    return _acados.ocp_nlp_set_model(self, f, step)
RuntimeError: Something went wrong when compiling the model.

    Start 6: ocp_qp_py
6/7 Test #6: ocp_qp_py .............................   Passed    0.16 sec
    Start 7: sim_py
7/7 Test #7: sim_py ................................***Failed    0.23 sec
cc: error: vde.c: No such file or directory
cc: fatal error: no input files
compilation terminated.
Traceback (most recent call last):
  File "/home/zw/acados0829/examples/python/sim.py", line 10, in <module>
    sim = sim_solver('rk', ode_fun, sim_options)
  File "/home/zw/acados0829/build/swig/acados.py", line 187, in __init__
    this = _acados.new_sim_solver(*args)
RuntimeError: Something went wrong when compiling the model.

71% tests passed, 2 tests failed out of 7

Total Test time (real) =   3.73 sec

The following tests FAILED:
Cannot create directory /home/zw/acados0829/build/Testing/Temporary
Cannot create log file: LastTestsFailed.log
	  5 - ocp_nlp_py (Failed)
	  7 - sim_py (Failed)
Errors while running CTest

roversch, 2017-08-29:
`vde.c` is autogenerated by `acados`. These are the model  equations of your ODE. 

The fact that it cannot be generated might have to do with access restrictions. Try 

    sudo chmod 777 acados

on your acados folder (the one you cloned).

You can also `make install` and try the example from a different folder where you have access to write to that folder (make sure your installation folder is in the PYTHONPATH).

On the other hand, maybe `vde.c` might have been generated in a different folder? Can you check that?

Flancis, 2017-08-29:
I did as you suggested but the error remains the same. Also, I didn't find any file named `vde.c`, I did find some files with names similar to `vde_chain_nm2.c` though.

roversch, 2017-08-29:
In `ocp_nlp.py`, can you try the following?

    ode_fun.generate('vde.c')

Does this generate anything?

Flancis, 2017-08-29:
Whenever I tried to run that file, it just gives the following error.

Traceback (most recent call last):
  File "examples/python/ocp_nlp.py", line 5, in <module>
    from acados import ocp_nlp, ocp_nlp_solver
ImportError: cannot import name 'ocp_nlp_solver'

roversch, 2017-08-29:
But when you did `ctest` it was able to import `ocp_nlp_solver`, right? (see above)

Can you add `ode_fun.generate('vde.c')` in `ocp_nlp.py` and run `ctest --output-on-failure` again please?

Otherwise, a new file with all the `acados` stuff removed will do as well.

Flancis, 2017-08-30:
Actually, the test of `ocp_nlp.py` never succeeded.

I added the line of code you mentioned, and the output is a little different from before. It says that `ImportError: No module named 'casadi'`. Please check below for details.

Test project /home/zw/acados0829/build
    Start 1: mass_spring_hpmpc_example
1/7 Test #1: mass_spring_hpmpc_example .............   Passed    1.34 sec
    Start 2: mass_spring_qpoases_example
2/7 Test #2: mass_spring_qpoases_example ...........   Passed    0.09 sec
    Start 3: chen_nmpc_qpoases_example
3/7 Test #3: chen_nmpc_qpoases_example .............   Passed    1.56 sec
    Start 4: pendulum_partial_tightening_example
4/7 Test #4: pendulum_partial_tightening_example ...   Passed    0.01 sec
    Start 5: ocp_nlp_py
5/7 Test #5: ocp_nlp_py ............................***Failed    0.40 sec
Traceback (most recent call last):
  File "/home/zw/acados0829/examples/python/ocp_nlp.py", line 6, in <module>
    from example_model import example_model
  File "/home/zw/acados0829/examples/python/example_model.py", line 1, in <module>
    from casadi import SX, Function, vertcat
ImportError: No module named 'casadi'

    Start 6: ocp_qp_py
6/7 Test #6: ocp_qp_py .............................   Passed    0.16 sec
    Start 7: sim_py
7/7 Test #7: sim_py ................................***Failed    0.19 sec
Traceback (most recent call last):
  File "/home/zw/acados0829/examples/python/sim.py", line 4, in <module>
    from example_model import example_model
  File "/home/zw/acados0829/examples/python/example_model.py", line 1, in <module>
    from casadi import SX, Function, vertcat
ImportError: No module named 'casadi'

71% tests passed, 2 tests failed out of 7

Total Test time (real) =   3.75 sec

The following tests FAILED:
	  5 - ocp_nlp_py (Failed)
	  7 - sim_py (Failed)
Errors while running CTest

roversch, 2017-09-02:
Wait, so you cannot import CasADi? But before it worked, no? 

> RuntimeError: Something went wrong when compiling the model. 

Can you try the following in a Python file in the home folder (e.g. ~/):

    from casadi import *
    x = SX.sym('x')

roversch, 2017-09-12:
@Flancis any progress with this? :)

Flancis, 2017-09-14:
Sorry for the late reply. The codes you mentioned worked in python idle.

roversch, 2017-09-14:
Also the `ocp_nlp.py` and `ocp_qp.py` examples of `acados`?

Flancis, 2017-09-15:
Unfortunately no. When I tried to run those two examples, the following error codes came out.

========== RESTART: /home/zw/acados0829/examples/python/ocp_nlp.py ==========
Traceback (most recent call last):
  File "/home/zw/acados0829/examples/python/ocp_nlp.py", line 5, in <module>
    from acados import ocp_nlp, ocp_nlp_solver
ImportError: cannot import name 'ocp_nlp_solver'

roversch, 2017-09-16:
Okay, so CasADi works. But somehow acados is sometimes found, sometimes not. Would you mind pasting your $PYTHONPATH here? You can get it by typing 

    echo $PYTHONPATH

at the terminal. Also

    ls ~/local/lib

would be helpful. Thanks!

Flancis, 2017-09-16:
Hey. The python path is as follows.
/home/zw/acados0829:/home/zw/acados0829/external/casadi-py35-np1.9.1-v3.1.1:

And the second one is
acados.py  _acados.so  libqpdunes.a  libqpOASES_e.a

roversch, 2017-09-16:
okay try this from the terminal: 

    export PYTHONPATH=~/local/lib:$PYTHONPATH

and run again.

It is advised to add that line at the bottom of your `~/.bashrc` file. It tells Python where to find acados (which is installed in `~/local/lib`)

Flancis, 2017-09-16:
I tried to run the file, but the error is the same.
Traceback (most recent call last):
  File "/home/zw/acados0829/examples/python/ocp_qp.py", line 3, in <module>
    from acados import ocp_qp, ocp_qp_solver
ImportError: cannot import name 'ocp_qp_solver'

roversch, 2017-09-16:
Maybe it's an IDLE setting. Let's try with the Python interpreter from the terminal.

Can you (after setting the PYTHONPATH) try

    python3 ocp_qp.py 

or just

    python ocp_qp.py 

from the /examples folder?

Flancis, 2017-09-16:
It's working now. For the nlp example, there should be a plot coming out, right? But the plot just flashed after running the code and then disappeared very soon. Is there anything wrong with matplotlib or it's a normal behavior?

roversch, 2017-09-16:
The plots are made in 'interactive' mode. This means the code continues to run after the plot is shown. My advise would be to use `ipython`, which is an interactive environment.

Run with `ipython`, in the interactive shell that opens type `run ocp_nlp.py`.

Otherwise, you can adapt the example script by removing `plt.ion()` and put a `plt.show()` at the end.

Great that it works now!

Flancis, 2017-09-16:
Yeah, it worked finally. Thanks for your detailed and quick explanation for such a long period. :)

roversch, 2017-09-16:
No worries :) Closing the issue.

closed 2017-09-16 by roversch

Flancis, 2017-09-16:
Hi again. I tried to install acados in another computer. Similar error happened again...
`from casadi import *`
I tried to run the above code and the following error came out.
from: can't read /var/mail/casadi

I also tried to run `python ocp_qp.py` and it worked fine.

Flancis, 2017-09-16:
And on that computer, I tried to run the ocp example in MATLAB interface. And the following error came out.
Warning: NARGCHK will be removed in a future release. Use NARGINCHK or NARGOUTCHK instead. 
> In import (line 3)
  In ocp_nlp_example (line 1) 
Error using import>locatepkg (line 31)
Package 'acados' cannot be located in the path

Error in import>import1 (line 18)
 pkgpath = locatepkg(pkgname_parts{1});

Error in import (line 6)
   import1(varargin{i});

Error in ocp_nlp_example (line 1)
import acados.*

roversch, 2017-09-16:
For Python:

> can't read `/var/mail/casadi`

I don't know where the `var/mail` comes from, but that is a folder used for e-mail and thus definitely wrong. Check where you installed CasADi (by default `acados/external/casadi-...`) and point the PYTHONPATH there, as before. 

For MATLAB:

> Package 'acados' cannot be located in the path

Same thing: add `~/local/lib` to your path (as well as CasADi), by right-clicking and choosing 'Add to path'

Flancis, 2017-09-16:
I did all those settings right after installing acados and also tried reinstalling the package. But the error is the same.

roversch, 2017-09-16:
is acados (for MATLAB) actually installed ? There should be a folder `+casadi` in `~/local/lib`.

Flancis, 2017-09-16:
The following files and folders are in `~/local/lib`.
acados.py   _acados.so   libqpdunes.a    __pycache__
acados.pyc  libacados.a  libqpOASES_e.a

So no `+casadi` in that folder.

roversch, 2017-09-16:
Sorry, I meant '+acados'. Try from your acados/build folder:

`cmake -D SWIG_MATLAB=ON ..`

`make install`

and then you should get a '+acados' folder.

Flancis, 2017-09-16:
You mean a `+acados` folder would be generated in the lib folder? After running those commands, still no such folder was found.

roversch, 2017-09-20:
Can you remove `build/CMakeCache.txt` and run CMake again please? Then paste the output here.

Flancis, 2017-09-20:
Hey, here is the output after running cmake.
cmake -D SWIG_MATLAB=ON ..
-- Build type is Release
-- Installation directory is /home/u64/local
-- MATLAB not found. Set MATLAB_ROOT
-- Will look for MATLAB libraries in 
-- MATLAB not found. If you do have it, set MATLAB_ROOT and reconfigure (missing:  MATLAB_ROOT 
MATLAB_INCLUDE_DIR MATLAB_LIBRARIES MATLAB_MEX_PATH MATLAB_MEXEXT_PATH MATLAB_MEX_EXT) 
-- MATLAB not found. Set MATLAB_ROOT
-- Will look for MATLAB libraries in 
-- MATLAB not found. If you do have it, set MATLAB_ROOT and reconfigure (missing:  MATLAB_ROOT MATLAB_INCLUDE_DIR MATLAB_LIBRARIES MATLAB_MEX_PATH MATLAB_MEXEXT_PATH MATLAB_MEX_EXT) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/u64/acados0916/build

roversch, 2017-09-20:
So it turns out MATLAB could not be located.

Could you run again with `-D MATLAB_ROOT=/path/to/matlab/` like CMake asks?
That should be a path to a folder like MATLAB_2015b or something similar.