acados/acados #119: Re-activate unit tests for QP solvers

dkouzoup, 2017-09-01:
Currently unit testing of QP solvers is deactivated (old order of bounds in read_ocp_qp, old qpoases interface, etc). The testing should be replaced asap in python.

rienq, 2017-09-01:
there is a different order of bounds?

roversch, 2017-09-01:
Yeah @giaf changed it back to [u, x] instead of [x, u] ..

I think it's a bit confusing because the natural way for me is [x, u],
but apparently it was not easy to swap the bounds again in hpipm.

rienq, 2017-09-01:
I agree, this is not intuitive at all and should not affect the solver performance so why this change exactly??

dkouzoup, 2017-09-02:
to what I recall the whole condensing code that is coming from hpipm is written in the [u x] order and it was somehow difficult to calculate the memory size required in a not messy way.. But @giaf can elaborate more on that and try to convince us once again :P Or perhaps a solution that would make everyone happy would be to use idxbx idxbu, lbx, lbu, ubx, ubu?

rienq, 2017-09-02:
Yea, my code broke because of this so it would have been nice to get a heads up.. ;-)

But I really don't agree with just switching the order from what everyone can agree is quite a natural choice to something rather awkward.. going for what @dkouzoup suggests, and splitting up the bounds for states and controls, is at least transparent and would therefore be already a lot better in my opinion..

dkouzoup, 2017-09-02:
Yeah we were too quick merging giaf's branch with the master before the hackathon and forgot this significant change that was developed in parallel there.. Since we have quite some people against it liI suggest that @giaf first explains again here what the problem was with condensing and if it cannot be fixed easily, we go with splitting the bounds (the more I think of it, I kind of like the second option better anyway).

giaf, 2017-09-02:
Ok, so the issue is in the Riccati recursion, or better its efficient implementation in HPMPC/HPIPM. If you take any of my papers, you see that variables are stacked as [u; x], as this exact order is necessary to implement efficiently the Riccati recursion by having a lot of operations performed implicitly by the definition of Cholesky factorization. So, forgetting for a second about conventions, this is the natural ordering of variables for this algorithm.
(Partial) condensing in HPIPM is implemented using an analogue scheme, so again [u; x] is the natural ordering for the algorithm.
In HPIMP the focus in on having a QP structure that is the most natural and efficient possible for the algorithm, so [u; x] is the variable ordering. If an historical convention prevents the efficient implementation of an algorithm, I don't see the problem in dropping it. That's progress ;)

The issue with ACADOS is in the routine to compute the memory size. This needs to call the corresponding routine in HPIPM, that expects the [u; x] ordering. Now, since the qp_in data is supposed to be const, I am not allowed to swap the indexes of u and x in place in idxb. Cast to a non-const is a bad hack. Then I need memory to allocate a temporary idxb_swapped. Dynamic or automatic memory allocation are not an option for performance/compatibility reasons. Static memory allocation is not possible since N is not known at compile time. What is left? Having another routine that asks how much memory the routine to allocate memory needs? :D It does not look very neat ;) And it is not even efficient to keep swapping idxb in both the routine to compute the memory size, and then in the interface each time the solver is called...

dkouzoup, 2017-09-03:
Thanks for the detailed answer @giaf! It seems to me in that case that splitting the bounds in x and u would not make this procedure any easier. Or would it? A naive question: if the problem is only in the calculate_size routine, could we not write a calculate_size_flipped_bounds that would do the same calculation assuming [x u] ordering? The conversions from the acados format to the blasfeo/HPIPM format have to take place online anyway since you stack matrices together and have a different matrix format, no?

roversch, 2017-09-03:
> "If an historical convention prevents the efficient implementation of an
algorithm, I don't see the problem in dropping it. That's progress ;)"

To me that sounds like bad interface design. A change in variables in a
lower level shouldn't bubble up all the way to the high-level interface.

giaf, 2017-09-03:
The what is your suggestion for a good interface design?

Sure it's possible to split idxb in idxbx+idxbu, and idxbs in idxsx+idxsu+idxsg, but this just creates a lot of more arguments... is it worth?

giaf, 2017-09-03:
Stacking the variables in the opposite ordering than used in the algorithm does not sound like the best option either, especially for a performance-aware piece of software.

rienq, 2017-09-03:
In the function to compute the memory size, you would not have to swap the indices right? You can easily compute how many control and state bounds you have by going through the vector in the unswapped/wrong order without problems.. only for your solver itself, you would maybe need a swapped version of idxb which just means you have to put it in workspace like all other solvers are putting things in workspace because the data format is not exactly how the solver wants it, right? I really don't see how this is any different from e.g. qpDUNES having to put the lower and upper bounds and the Hessian etc in workspace because it expects a different format..

The fact that you say that this is the "natural ordering for this algorithm" points out to me why the current situation is not ideal, because we are not looking for an ordering that is slightly better for one specific algorithm, we are designing an interface that is as close as possible to the most natural optimal control problem formulation and forms a generalization of the "natural interfaces" of many different existing algorithms as well as algorithms that still need to be invented in the future..
So if, tomorrow, you come up with an algorithm that can be better for yet another problem structure, e.g., moving horizon estimation problems, would you want to switch the order of the bounds again or create two different versions of our ocp_qp interface?

It is clear that the interface to all solvers is making some kind of sacrifice because the ocp_qp interface is not exactly putting the data directly into the solver's format but isn't that exactly the whole point of acados? To make a large range of solvers available under a common interface? It is our (challenging) task to find out which sacrifices are worth it and which ones are not but, in my opinion, this particular one of changing the order of the bound indices just for one particular algorithm is not worth it at all..

If you want to squeeze out the last 1-5% of performance, then you should hardcode your controller using directly all the solver components you need and not go through our flexible environment in acados. The point is that we know, from experience with ACADO for example, that most people are more interested in flexibility and user-friendliness than performance, both in academia and industry, such that we can sacrifice minor performance issues for a better and more intuitive common interface.. right?

roversch, 2017-09-03:
> "we are designing an interface that is as close as possible to the most
natural optimal control problem formulation and forms a generalization of
the "natural interfaces" of many different existing algorithms as well as
algorithms that still need to be invented in the future.."

I can only agree to that! It indeed seems weird to change the ordering in
our interface just for the sake of 1 particular solver. Having a clearly
defined 'natural' interface to start from would possibly prevent changing
the ordering again (and again and again...)

What about the option of assuming there are always bounds on all variables,
and populate the unconstrained ones with +/-inf and then eliminate them
before we call any of the low-level solvers?

rienq, 2017-09-03:
Something more related to the original topic of this issue: is it correct that currently all unit tests are disabled? Travis seems not to be running the ocp_nlp and sim unit tests either..

giaf, 2017-09-03:
Ok, fine, if you say that the primal goal of ACADOS is a flexible interface that can accommodate many different existing algorithms as well as algorithms that sill need to be invented in the future, and performance is not that critical, that is perfectly fine for me.

But then, in my opinion, you should not make ACADOS dependent on BLASFEO and HPIPM in the first place. The situation as it is now is something in between: ACADOS depends on both BLASFEO and HPIPM (only qpDUNES can be used without HPIPM, and implicit integrators use LA from BLASFEO). But their full potential is not exploited.

In my opinion, it should either one or the other. Or it is decided that ACADOS is independent of them, and then it should be possible have a fully working library without using them. Or it is decided that it depends on them, and then they should be exploited fully.

Up to you.

rienq, 2017-09-03:
You are right that acados currently benefits a lot from BLASFEO and HPIPM, because we all agree that both those projects are amazing!! Isn't the point of BLASFEO to make its linear algebra routines more easily available for projects like acados? Such that we can benefit from high-performance kernels but we can also switch back to reference implementations that are more portable or even use other BLAS libraries whenever needed.. I don't see why you are making it sound like a binary decision between performance or flexibility: obviously, we want both but there will always be a trade-off between whether you pay more in performance or in flexibility..

But its not true that all other solvers depend on them.. the implicit integrators for example still have the option to use the simple triple loop implementation whenever the blasfeo libraries would not be available on the computing platform for whatever reason.

For QP solvers it is even more obvious: qpDUNES, OOQP, qpOASES and hopefully more in the future, can and should all be able to be used without HPIPM and BLASFEO.. 

The only reason that you say that qpOASES currently depends on HPIPM is that we only support condensing_qpoases as a combined option for sparse QPs but this is not a good situation in my opinion either: condensing and solving dense QPs could and probably should be separated.. 
From a mathematical point of view, condensing just takes an ocp_qp in and put an ocp_qp (of different dimensions) out such that you can pass it to qpOASES or a first order method or whatever.. there is no reason that qpOASES would need BLASFEO/HPIPM to solve a dense QP. So in my opinion, there should also be an interface to qpOASES that does not depend on anything else than qpOASES but that is yet another issue.

giaf, 2017-09-03:
BLASFEO and HPIPM are there, you decide how to use them. I myself would like to see what is their full potential and want to investigate that.

Getting back to a more practical point, can someone try to interface the condensing in HPIPM with the ACADOS framework, fulfilling the following constraints:
- the variables order is [x; u]
- the interface can not use malloc or variable size arrays
- the members of the qp_in interface are defined as const, and therefore they can not be edited
I failed in that, maybe you find a better solution.

Hint: in condensing, box constraints on states at stages 1,.. N-1 becomes general constraints, box constraints on inputs remain box constraints. So you need to tell the number of state and input box constraints before lb and ub are available.

giaf, 2017-09-03:
Before too many people write me worried, I make it clear: I gave up. I don't want to steer the ACADOS development. You decide how ACADOS should look like, and how you want to use BLASFEO and HPIPM.

For my personal interest, I want to see what is possible to do in the BLASFEO/HPIPM framework, but this is unrelated to the ACADOS development and doesn't want to affect it.

dkouzoup, 2017-09-04:
Hey guys, a few side notes: 

- @rienq, it might be useful indeed to have a condensing interface also as a separate entity at some point (if based on HPIPM it would only be a matter of a few lines) but I still think that the coupled interfaces should also be there, simply because it is not efficient to convert from column major to panel major, then back to column major and finally to row major (qpOASES) or panel major (HPIPM).h

- @roversh, the inf bounds although it would be convenient from the user's perspective, it sounds a bit dangerous in my opinion. We would need to be careful on what 'inf' is in all languages (python, matlab, c) and I think it doesn't solve the problem of the ordering either, right? 

- @giaf, lets join forces this week to find a clean solution for this issue ;)

dkouzoup, 2017-09-04:
PS: I guess the majority has spoken, I am working on reverting the order of bounds (leaving a flag FLIP_BOUNDS for now to support both cases in a transition period..).

roversch, 2017-09-04:
@dkouzoup Thanks a lot for taking the initiative here. 

One note about Inf/NaN: those are standardized by IEEE 754, and I don't think it's dangerous at all. We could use the `INFINITY` macro along with `isfinite()` function from the C standard library (`math.h`)

giaf, 2017-09-04:
I see a couple of dangers:
- they are introduced in C with the C99 standard.
- many QP solvers are not designed to handle them, as e.g. qpOASES, HPMPC, HPIPM.

rienq, 2017-09-05:
Hmm, about the condensing:
The problem with this is that you would have a lot of duplicate code in that case, which is really bad..
Instead, you could split up the condensing from the dense solvers and give the condensing module some options to spit out the matrices in a different storage format maybe for efficiency?

dkouzoup, 2017-09-05:
It would be really neat if we could give as an (optional) input to the condensing module the QP solver to follow such that the data are generated in the right format but I cannot think at the moment a clean way to pass this data from the one module to the other given our fixed format (input, output, args, memory, workspace) of each module. We should also let somehow the solver "know" that the input data are not in qp_in but somewhere else.. 

Regarding the order of bounds I was quite of nutral on this issue. @giaf convinced me at the time that swapping was the cleaner way to interface with his software but we are currently working on an idea how to overcome this issue, let's see. 

Finally as a side note: in the OOQP interface I realized (after some quite painful debugging process) that the solver returns garbage if the inf is set too large, e.g. 1e12, and works fine if it's 1e10... (Had to add those infs since bounds are double-sided).

giaf, 2017-09-05:
Just a small note. I do not want to reopen discussions and issue, I assure you I had more than enough. I just want to humbly make you notice that QPs and routines in HPIPM provide exactly this systematic, efficient and clean framework to combine OCP QP solvers with dense QP solvers with (partial) condensing routines that you are looking for now.

And that, in my humble opinion, there should be a clear separation between the font-end (designed to provide powerful and intuitive interfaces to the user) and the back-end (designed to be efficient and simplify the life of the poor developers).

dkouzoup, 2017-09-06:
fixed with #123

closed 2017-09-06 by dkouzoup