acados/acados #120: Segmentation fault in mass_spring example with qpOASES

rienq, 2017-09-01:
Starting program: acados/build/examples/c/mass_spring_qpoases_example 
 Test problem: mass-spring system with 4 masses and 3 controls.

 MPC problem size: 8 states, 3 inputs, 15 horizon length, 11 two-sided box constraints, 0 two-sided general constraints.

work space size: 0 bytes
memory: 7627072 bytes

Program received signal SIGSEGV, Segmentation fault.
QProblem_setupAuxiliaryQPsolution (_THIS=_THIS@entry=0x7ffff7002540, xOpt=xOpt@entry=0x0, yOpt=yOpt@entry=0x7ffff6fcd274)
    at acados/external/qpOASES/src/QProblem.c:3429
3429					_THIS->y[i] = yOpt[i];
(gdb) bt
#0  QProblem_setupAuxiliaryQPsolution (_THIS=_THIS@entry=0x7ffff7002540, xOpt=xOpt@entry=0x0, yOpt=yOpt@entry=0x7ffff6fcd274)
    at acados/external/qpOASES/src/QProblem.c:3429
#1  0x00000000004311cb in QProblem_solveInitialQP (_THIS=_THIS@entry=0x7ffff7002540, xOpt=xOpt@entry=0x0, 
    yOpt=yOpt@entry=0x7ffff6fcd274, guessedBounds=guessedBounds@entry=0x0, guessedConstraints=guessedConstraints@entry=0x0, 
    _R=_R@entry=0x0, nWSR=0x7fffffffce5c, cputime=0x7fffffffce60) at acados/external/qpOASES/src/QProblem.c:2299
#2  0x00000000004322cb in QProblem_initW (_THIS=_THIS@entry=0x7ffff7002540, _H=_H@entry=0x7ffff6fbd3e8, _g=_g@entry=0x7ffff6fcc190, 
    _A=_A@entry=0x7ffff6fc1330, _lb=_lb@entry=0x7ffff6fcc5c8, _ub=0x7fff00000000, _ub@entry=0x7ffff6fcc730, _lbA=0x7fffffffce5c, 
    _ubA=0x7fffffffce60, nWSR=0x7fffffffce5c, cputime=0x7fffffffce60, xOpt=0x0, yOpt=0x7ffff6fcd274, guessedBounds=0x0, 
    guessedConstraints=0x0, _R=0x0) at acados/external/qpOASES/src/QProblem.c:465
#3  0x000000000040d290 in ocp_qp_condensing_qpoases (qp_in=qp_in@entry=0x7fffffffd040, qp_out=qp_out@entry=0x7fffffffd010, 
    args_=args_@entry=0x7fffffffd000, memory_=memory_@entry=0x7fffffffdb60, workspace_=workspace_@entry=0x67f740)
    at acados/acados/ocp_qp/ocp_qp_condensing_qpoases.c:446
#4  0x0000000000409af4 in main () at acados/examples/c/mass_spring_qpoases.c:746

dkouzoup, 2017-09-02:
ups.. we thought we magically fixed this issue with @giaf but I think the example was just deactivated.. Do you get this error only on travis or also on your computer? Because we could not reproduce it locally. One temp solution is to change initW to init in the qposes interface, but we have to track down the problem at some point

rienq, 2017-09-02:
I see, this was on my computer.. I have the impression that Travis is not actually running the examples anymore, is it?

dkouzoup, 2017-09-02:
no travis is running the examples (ctest) but not this specific one. That's great that your seg fault happens locally. We could not reproduce it on any of our laptops! Can

rienq, 2017-09-03:
Hmm, replacing initW by init did solve the above segfault.. but now another segfault appears when running the same example in BLASFEO, from gdb:

(gdb) run
Starting program: acados/build/examples/c/mass_spring_qpoases_example 
 Test problem: mass-spring system with 4 masses and 3 controls.

 MPC problem size: 8 states, 3 inputs, 15 horizon length, 11 two-sided box constraints, 0 two-sided general constraints.

work space size: 0 bytes
memory: 7627072 bytes

Program received signal SIGSEGV, Segmentation fault.
0x0000000000440136 in d_cvt_vec2strvec ()
(gdb) bt
#0  0x0000000000440136 in d_cvt_vec2strvec ()
#1  0x000000000040d2fc in ocp_qp_condensing_qpoases (qp_in=qp_in@entry=0x7fffffffd040, qp_out=qp_out@entry=0x7fffffffd010, 
    args_=args_@entry=0x7fffffffd000, memory_=memory_@entry=0x7fffffffdb60, workspace_=workspace_@entry=0x67f740)
    at acados/acados/ocp_qp/ocp_qp_condensing_qpoases.c:476
#2  0x0000000000409af4 in main () at acados/examples/c/mass_spring_qpoases.c:746

roversch, 2017-09-03:
Try compiling BLASFEO with -O2 instead of -O3, you can do this by putting

set(CMAKE_C_FLAGS_RELEASE "")

rienq, 2017-09-03:
Hmm, that's interesting, the example runs now after recompiling acados in debug mode which removes all optimization flags I guess..
However, why does that make a difference? Plus, I have code similar to the mass_spring_qpoases example which links to the generated acados libraries which still throws the same segmentation fault even after recompiling in debug mode:

Program received signal SIGSEGV, Segmentation fault.
0x0000000000443b4f in d_cvt_vec2strvec (m=30, a=0x7ffff6fdf598, sa=0x7ffff6ff2c00, ai=0) at acados/external/blasfeo/auxiliary/d_aux_lib4.c:2455
2455			pa[ii] = a[ii];
(gdb) bt
#0  0x0000000000443b4f in d_cvt_vec2strvec (m=30, a=0x7ffff6fdf598, sa=0x7ffff6ff2c00, ai=0) at acados/external/blasfeo/auxiliary/d_aux_lib4.c:2455
#1  0x00000000004059c1 in ocp_qp_condensing_qpoases (qp_in=0x7fffffffd9c0, qp_out=0x7fffffffd990, args_=0x7fffffffd960, memory_=0x7fffffffda60, workspace_=0x6ba850) at acados/acados/ocp_qp/ocp_qp_condensing_qpoases.c:476
#2  0x0000000000401d73 in main (argc=4, argv=0x7fffffffdd18) at chain_MPC.c:426

giaf, 2017-09-04:
In the mass_sping_qpoases example, I tracked the bug to most likely be a gcc bug that happens when BLASFEO is compiled with -O3. With -O3, gcc vectorizes the code, and it uses aligned loads. If the memory address of the load is not properly aligned, you get segmenation fault. And apparently gcc is not checking for the alignment.

I am not 100% sure of the source of the bug. The final verification would be to hand fix the generated assembly code with a non-aligned load, assembly the corrected .s into a .o, and use this in the ACADOS library.

Note that by default BLASFEO is compiled with -O2 as stand alone, so this happens only when ACADOS overwrites the default "suggested" BLASFEO compilation flags.

dkouzoup, 2017-09-04:
I had the same problem with the create_ocp_qp_in implementation on my fork. In this case, gcc+O3 (doesn't happen with clang or gcc with O2) somehow messes up the memory that is set up by the allocate+assign routines. @giaf we can have a look at the assembly code together and see if it's the same source of problems. @rienq Can you tell us some info on your operating system and compiler where the initW seg fault happens?

giaf, 2017-09-04:
sure

rienq, 2017-09-05:
I am still having the same issue actually: replacing -O3 with -O2 worked for the mass spring example, but then I still get the exact same segfault whenever I use the resulting static library in my own example code, anyone experienced that before?
I am running gcc 5.4.0 on Ubuntu 16.04 by the way

roversch, 2017-09-05:
A small note from my side: the C99 (also ANSI C) standard says that types have _alignment requirements_, i.e. we cannot put `double`s wherever we want. Furthermore, this property is implementation specific, so GCC handles it differently from Clang, etc.

See also https://stackoverflow.com/questions/11108328/double-alignment#11110283 

So, after debugging a bit, I found that our `double`s are sometimes not aligned on 64-bit boundaries, hence the segfault (more specifically, a general protection fault or EXC_I386_GPFLT). I assume here that `-O3` removes checking facilities preventing these segfaults from happening.

giaf, 2017-09-05:
This is not true. 

On x64_64, doubles do not have to be aligned to any particular boundary, see e.g. the instruction
https://software.intel.com/sites/landingpage/IntrinsicsGuide/#text=load_sd&expand=3080 
For performance reasons, it is (or better, it was, on older architectures) preferable them to be aligned to 64-bit boundaries.

What, in my humble opinion, is happening in the code, is that it is performing an aligned vector load on an unaligned data
https://software.intel.com/sites/landingpage/IntrinsicsGuide/#text=load_pd&expand=3080,3060
causing the segmentation fault.

The segmentation fault does not happen with -O2 because in that case the code only uses movsd (scalar load/store), that does not require any particular data alignment.

giaf, 2017-09-06:
But is it true that, if the doubles are not aligned to 64-bit boundaries, then no vector code like the one produced by gcc could ever work.

In the ACADOS workspace framework, this can be done as it is done in HPIPM.
A little extra memory is allocated for each structure
https://github.com/giaf/hpipm/blob/master/ocp_qp/x_ocp_qp.c#L74
for double-word alignment, 8 bytes is enough.
It may make sense to make the memory size of the entire structure multiple of some nice number of bytes, like 64 (an entire typical cache line) in HPIPM
https://github.com/giaf/hpipm/blob/master/ocp_qp/x_ocp_qp.c#L73

In the "create struct" routine, usually the pointer are allocated first. In a 32-bit OS, pointers are 4-bytes aligned, therefore after the pointers allocation and before the doubles allocation, an explicit alignment is performed
https://github.com/giaf/hpipm/blob/master/ocp_qp/x_ocp_qp.c#L213
Again, for double-word alignment 64 has to be replaced with 8.

Have fun.

dkouzoup, 2017-09-06:
Thanks for the investigation guys! I will try it out on my branch where I had the same segfaults happening for the memory allocated for qp_in and get back to you

roversch, 2017-09-06:
@giaf Thanks for that reference. This is not a question of architecture, this is a question of C standard requirements. GCC is allowed to produce optimized code that assumes that our doubles are 64-bit aligned, and if we do not abide, we get a segfault.

dkouzoup, 2017-09-06:
The alignment think did fix my problem with gcc+O3, thanks @roversch and @giaf

roversch, 2017-09-13:
This commit 9b64d8844b0805a99e7e60c3f0b699b3326ef7e8 (hopefully) fixes this. Will reopen when this occurs again.

closed 2017-09-13 by roversch

LilHu7, 2023-08-16:
Hi,
I also encountered similar issues, segmentation fault (core dumped). 
That happens when i try to run acados_solve(acados_ocp_capsule), and the gdb reports "received signal SIGSEGV, Segmentation fault" in sgemm() from acados/lib/libqpOASES_e.so.3.1.

I try to change the g++ -o3 to o2, but it not works.
@giaf @dkouzoup  can you provide some solutions?