acados/acados #125: Dependency on "d_eye" in libhpmpc.a

rienq, 2017-09-06:
When compiling code that uses the condensing_qpoases solver and linking against 
*libacados.a libqpOASES_e.a libhpipm.a libblasfeo.a*
you get following error messages:

`libacados.a(math.c.o): In function `padeapprox':
/home/quirynen/Software/acados/acados/utils/math.c:368: undefined reference to `d_eye'
/home/quirynen/Software/acados/acados/utils/math.c:398: undefined reference to `d_eye'
/home/quirynen/Software/acados/acados/utils/math.c:434: undefined reference to `d_eye'
/home/quirynen/Software/acados/acados/utils/math.c:485: undefined reference to `d_eye'
/home/quirynen/Software/acados/acados/utils/math.c:554: undefined reference to `d_eye'`

which can be avoided by linking additionally against *libhpmpc.a*, but this maybe seems unnecessary to depend on hpmpc in this case?

roversch, 2017-09-14:
I agree, @giaf do you want to have `d_eye` in blasfeo maybe?

dkouzoup, 2017-09-14:
otherwise we just code it and add it in math.c (with another name to avoid  confusion)

FreyJo, 2020-10-07:
Closing, these `d_eye`s are not in the acados repository anymore.

closed 2020-10-07 by FreyJo