acados/acados #134: Bug in ocp_qp_condensing_qpoases in combination with pyplot

roversch, 2017-09-26:
This is a weird bug: if you run `qp_solver.evaluate` or `nlp_solver.evaluate` with `condensing_qpoases`, qpOASES fails with code 36, because NaNs appear in some places (off-diagonal in the Hessian for example).

Without plotting, no problem. With other qp solvers, no problem.

roversch, 2017-09-27:
Found the deeper cause: apparently `dgemm_nn_libstr` is not robust against NaNs. For example, if we initialize `qpd->Hg` with garbage, the garbage gets overwritten by the condensing (i.e. `d_cond_RSQrq_N2nx3`). Somehow, NaNs are not overwritten.

I think the NaNs originate from pyplot producing them in the memory which is reused by acados.

Can you please comment on this @giaf ?

roversch, 2017-09-27:
This also happens in hpmpc and hpipm, by the way

zanellia, 2017-09-27:
Isn't that how a dgemm works? (see
http://www.netlib.org/lapack/explore-html/d1/d54/group__double__blas__level3_gaeda3cbd99c8fb834a60a6412878226e1.html#gaeda3cbd99c8fb834a60a6412878226e1)
If C is initialized with NaNs, you should get NaNs in the output.

roversch, 2017-09-27:
Thanks for that reference! From this, I conclude we should always zero the
memory before calling anything that calls dgemm.

dkouzoup, 2017-09-27:
Good point Andrea! I would opt for using malloc and then zeroing the _data_ inside the assign function rather than using calloc at the initialization point (which may eventually hide valgrind read-uninitialized-memory errors..)

roversch, 2017-09-28:
@giaf I guess this should go into `create_dense_qp` in hpipm?

dkouzoup, 2017-09-28:
I suppose that @giaf never encountered this as he is using d_zeros to initialize the memory. It would be however more robust to explicitly add zeros in the parts where NaNs cannot be overwritten.. Create is happening offline anyway.

closed 2020-10-08 by FreyJo