acados/acados #138: qpOASES interface segfaults for certain problem dimensions

rienq, 2017-09-30:
This is a well-known issue by most of us, but I think it should be addressed at some point for two important reasons:
1) Someone's code might segfault simply because of this hard-coding:
https://github.com/acados/acados/blob/master/external/qpOASES/include/qpOASES_e/Constants.h#L62-L63
which can be a pain to find out and therefore someone might lose a lot of time over this.
2) Another thing is that currently the computation times for the qpOASES interface, in my experience, are completely unrepresentative for the actual capabilities of qpOASES because of this hard-coding of the memory. Obviously, qpOASES will perform the best when you actually set the number of variables and constraints to the dimensions that you have as is done for example in ACADO (code generation). By fixing these constants to a large value and then solving a range of problems with different dimensions, the computation times for the smaller problems are much (much!) larger than they should be! This is observed more when using acados on an embedded ARM processor, compared to for example your high-end personal computer of course..

dkouzoup, 2017-10-02:
I completely agree. The issue has actually high priority in our todo list, thanks for pointing it out ;)

roversch, 2017-10-02:
I agree. I recently found, for smaller problems, the loss accounted for the
'wrong' dimensions might be up to 50% of the computation time.

closed 2017-10-27 by dkouzoup