acados/acados #140: ocp_qp args-dependent memory allocation in ocp_nlp_gn_sqp_create_memory

zanellia, 2017-09-30:
Memory allocation for the ocp_qp module in ocp_nlp_gn_sqp_create_memory cannot have ocp_qp args as input, since args for the qp solver are created inside ocp_nlp_gn_sqp_create_memory. However, in some cases, memory allocation for the qp_solver depends on them, e.g. partial tightening.

zanellia, 2017-09-30:
A possible solution would be to have qp options as structs instead. Any thoughts?

dkouzoup, 2017-10-02:
> A possible solution would be to have qp options as structs instead.
Aren't the qp options structs already? I think the best way to handle this is the qp options struct to be a field in the nlp options struct

FreyJo, 2020-04-20:
I think this is resolved, since we have:
```
    int (*memory_calculate_size)(void *dims, void *opts);
    void *(*memory_assign)(void *dims, void *opts, void *raw_memory);
```
https://github.com/acados/acados/blob/aa904981ee462c64307afe40b03b0f4211d6e0a9/acados/ocp_qp/ocp_qp_common.h#L93

closed 2020-04-20 by FreyJo