acados/acados #148: valgrind reports conditional jumps in nonlinear_chain_sqp_example

zanellia, 2017-10-17:
valgrind reports conditional jumps in nonlinear_chain_sqp_example and nonlinear_chain_ocp_nlp_example on master branch after Big Merge

The ouput below:

`==15504== Memcheck, a memory error detector
==15504== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==15504== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==15504== Command: ./nonlinear_chain_ocp_nlp_example
==15504== 

----- NUMBER OF FREE MASSES = 1, d = 2 (Exact Newton) -----
==15504== Conditional jump or move depends on uninitialised value(s)
==15504==    at 0x548D72: d_compute_qp_size_ocp2dense_rev (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==    by 0x538780: ocp_qp_condensing_qpoases_calculate_memory_size (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==    by 0x538CD0: ocp_qp_condensing_qpoases_create_memory (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==    by 0x539AD0: ocp_qp_condensing_qpoases_initialize (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==    by 0x5373C1: create_ocp_qp_solver (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==    by 0x536A66: ocp_nlp_gn_sqp_create_memory (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==    by 0x41C297: main (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==  Uninitialised value was created by a heap allocation
==15504==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15504==    by 0x536F76: create_ocp_qp_in (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==    by 0x5369F9: ocp_nlp_gn_sqp_create_memory (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
==15504==    by 0x41C297: main (in /home/andrea/acados_temp/acados/build/examples/c/nonlinear_chain_ocp_nlp_example)
`

zanellia, 2017-10-17:
I might have a look into it later tonight, but I will be working on and off these days.

zanellia, 2017-10-19:
@dkouzoup @roversch I am not the most up to date on the latest qp interface. Since we seemed to have problems with initialization every now and then, would it make sense to initialize the memory to zeros by default in assign_memory?

roversch, 2017-10-20:
We could do that; however, that might conceal some valgrind errors.

Does initializing with zeros fix your problem?

dkouzoup, 2017-10-24:
Hey @zanellia, if you have a look here:

https://github.com/acados/acados/blob/master/acados/ocp_qp/ocp_qp_common.c#L217

I decided to go for initialization to zero of data only, not pointers, since some might not be initialized in the examples (e.g. matrix S or nondiagonal elements of Q). However as @roversch pointed out, setting everything to zero has in the past concealed underlying bugs and I find it a bit dangerous.

FreyJo, 2019-09-10:
just tried it.
The issue seems to be solved somewhen in the last 2 years :sweat_smile:  :tada:

closed 2019-09-10 by FreyJo