acados/acados #149: Python example not working on Ubuntu 16.04 - CMakeList definition adapation needed

schoelst, 2017-10-18:
In order to get the python examples to work I had to change line 30 of the top-level `CMakeList.txt` to: 

`add_definitions(-D_GLIBCXX_USE_CXX11_ABI=0)`

this is a known issue, this report is just to keep track of it and in case other people might have the same problem.

The error I received was:

```
Traceback (most recent call last):
  File "examples/python/ocp_nlp.py", line 5, in <module>
    from acados import ocp_nlp, ocp_nlp_solver
  File "/home/sot7rng/local/lib/acados.py", line 21, in <module>
    _acados = swig_import_helper()
  File "~/local/lib/acados.py", line 20, in swig_import_helper
    return importlib.import_module('_acados')
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: ~/local/lib/_acados.so: undefined symbol: _ZNK6casadi8Function4nameB5cxx11Ev
```

The fix proposed above works for me on Ubuntu 16.04. (According to the developers it has to be 1 for Ubuntu 14.04)

have a look at this stackoverflow issue for more detail:
https://stackoverflow.com/questions/33394934/converting-std-cxx11string-to-stdstring

jgillis, 2017-10-21:
Is this caused by the CasADi binaries shipping with the older CXX ABI?

FreyJo, 2019-08-07:
https://github.com/acados/acados/pull/419

closed 2019-08-07 by FreyJo