acados/acados #152: Why is the QP saved in the SQP options?

dkouzoup, 2017-10-27:
if I understand correctly from this:

https://github.com/acados/acados/blob/b6e92f5daf940e213f66766fc009eb239e4e65af/acados/ocp_nlp/ocp_nlp_sqp.c#L44

the QP is stored in the nlp options which does not make much sense to me. Why isn't in the memory?

dkouzoup, 2017-10-27:
ping @nielsvd , @roversch

nielsvd, 2017-10-27:
In case the QP Solver is stored in the SQP memory-struct there is no direct way to choose the QP solver. The previous implementation used a "work-around" by expecting the name of the QP solver in the SQP options-struct and passed the name to the QP Solver factory routine. This approach is to my opinion not to-the-point. What the SQP method needs is a QP solver, not a name of a QP solver. Instead, why not let the user call the QP Solver factory (or build the struct manually if preferred) and give the SQP routine what it actually needs. I believe this leads to more concise code, especially considering there may be other similar "options" you could potentially select via strings.

I'd like to stress I'm not against "string"-based selection of a QP solver, but this should not be in the core of acados. But rather in a different layer, with e.g., an NLP solver factory, which purpose is to greatly simplify the overall construction of an NLP solver.

Btw, the QP solver is in SQP options, not in NLP options.

dkouzoup, 2017-10-27:
Yes I agree that the previous solution was not ideal (was also a workaround) but I am still not sure whether the memory of the QP solver for instance should reside in the arguments of the SQP solver. Ideally, the SQP args should have a nested QP args struct, defining all the options for the QP solver while the rest would be in the memory, no?

nielsvd, 2017-10-27:
What you suggest is actually partially achieved in the current implementation, the args of the QP solver are in the SQP args (but QP memory and workspace are also). But I see it would be cleaner if the memory of QP solver is nested in the SQP memory, and similarly for the workspace. However, I don't see yet how this can be cleanly implemented.

nielsvd, 2017-10-27:
Actually, I think it should be possible. We could remove the ``memory`` and ``workspace`` fields from the ``ocp_qp_solver``-struct. However, I don't have a clear view of all implications yet.

dkouzoup, 2017-10-27:
Don't worry about it, Gianluca implemented it like this in hpipm actually and he is working towards a re-implementation in acados anyway.

closed 2018-03-29 by giaf