acados/acados #153: PR #147 broke matlab interface of condensing+qpoases (only for clang)

dkouzoup, 2017-10-29:
The matlab example on travis is hanging. The issue did not appear before the merge since no matlab examples are executed on PR.

dkouzoup, 2017-10-29:
@nielsvd

bnovoselnik, 2017-10-29:
Also, Matlab tests that do pass take unusually long time, around 30 sec, while in Python it's no more than 1 sec. But, that's also true for earlier commits as well. Is this normal?

roversch, 2017-10-29:
@bnovoselnik: Yes, that's normal. MATLAB is slow :p

As for the hanging: strange, only for one compiler. Will investigate soon

nielsvd, 2017-10-30:
I don't have a mac on which I can test this. I can have a look on Thursday, when I'm in Freiburg, what I've changed to cause this.

nielsvd, 2017-10-30:
No problems with clang+Matlab on my Linux installation. 
```
clang --version
clang version 5.0.0 (tags/RELEASE_500/final)
Target: x86_64-unknown-linux-gnu
Thread model: posix
```
To make sure we're talking about the same code. I run (modified from ``examples/matlab/ocp_qp_example.m``):
```matlab
import acados.*

qp = ocp_qp(struct('N',5, 'nx',2, 'nu', 1));

% specify OCP
qp.A = [0 0; 1 0];
qp.B = [0; 1];
qp.Q = eye(2);
qp.R = 1;

% specify initial condition
x0 = [1; 1];
qp.lb{1} = x0;
qp.ub{1} = x0;

% solve QP
solver = ocp_qp_solver('condensing_qpoases', qp);
output = solver.evaluate();
assert(abs(-0.5 - output.controls{1}) < 1e-8)
disp(output.states)
```

bnovoselnik, 2017-10-30:
@nielsvd: I think the other example (ocp_nlp) is causing trouble.

roversch, 2017-10-30:
Weird, this latest build output tells me MATLAB just hangs at startup..

https://travis-ci.org/acados/acados/jobs/294770601

dkouzoup, 2017-10-30:
@nielsvd we could reproduce this locally on a linux machine. Try the ocp_nlp example. If it still works, try the same clang version as travis does

FreyJo, 2019-08-07:
https://github.com/acados/acados/pull/419

closed 2019-08-07 by FreyJo