acados/acados #155: C++ wrapper

nielsvd, 2017-11-02:
It is relatively straightforward to write a minimalistic C++ wrapper the way acados deals with memory management now. A c++ brings two main advantages:
* The OO-framework can be used to sacrifice portability in favor of user friendliness effectively. Such as automatically memory allocation/freeing in the con-/destructors. And provide different methods of configuring the low-level routines, e.g., selecting the QP solver for a SQP solver using strings.
* It should greatly simplify the SWIG interface and its maintainability.

closed 2018-04-25 by dkouzoup

manucalop, 2021-07-04:
Was this implemented?

FreyJo, 2021-07-06:
Hi @manucalop
We have a stable C interface https://github.com/acados/acados/blob/master/interfaces/acados_c/
The Python and Matlab interfaces are based on that.
There was a design choice to build those interfaces directly in those languages instead of the C++ with SWIG approach, mentioned above.
Note that the problem specific C code that can be generated via Python/Matlab can be incorporated in a C++ framework relatively easily.

manucalop, 2021-07-06:
Thanks! For someone who just got here, we continued the discussion here: https://discourse.acados.org/t/acados-c-interface/471/2