acados/acados #161: python ocp_nlp.py example fails

schoelst, 2017-11-20:
Hi,

for me the python `ocp_nlp.py` example fails

```
$ python3 ocp_nlp.py
Traceback (most recent call last):
  File "ocp_nlp.py", line 15, in <module>
    nlp.set_model(ode_fun, step)
  File "/home/user/local/lib/acados.py", line 426, in set_model
    return _acados.ocp_nlp_set_model(self, f, step)
RuntimeError: Loading of vde_chen1.so failed. Error message: vde_chen1.so: cannot open shared object file: No such file or directory
```

I get the same error when I call it from the acaods root directory.

schoelst, 2017-11-20:
found the the problem:

in `swig/acados.i` line 210: it should read 
```
std::string path_to_library = std::string("./") + library_name;
```
instead of
```
std::string path_to_library = library_name;
```
Apparently this was changed 10 days ago, as part of some Windows bug fixes (244c1cfe62299e8c286187727b956550d0337d7b).

Here is the git diff output, to patch it:
```
diff --git a/swig/acados.i b/swig/acados.i
index da058ba..7e803ea 100644
--- a/swig/acados.i
+++ b/swig/acados.i
@@ -207,7 +207,7 @@ enum generation_mode {

 static casadi_function_t compile_and_load(std::string name, void **handle) {
     std::string library_name = name + std::to_string(global_library_counter++) + std::string(".so");
-    std::string path_to_library = library_name;
+    std::string path_to_library = std::string("./") + library_name;
     char command[MAX_STR_LEN];
     snprintf(command, sizeof(command), "%s -fPIC -shared -g %s.c -o %s", compiler, name.c_str(),
         library_name.c_str());
```

roversch, 2017-11-21:
Weird, we test the Python examples on Travis and it passes there. I pushed the fix that you propose, could you check if it solves your problem?

schoelst, 2017-11-22:
It works for me now, thanks!
Not sure if it has an influence, but I am on Ubuntu 16.04 and had to edit the top level CMakeList.txt by adding `add_definitions(-D_GLIBCXX_USE_CXX11_ABI=1)`. (This is also reported [here](https://github.com/acados/acados/issues/149) )

schoelst, 2018-01-18:
The issue with this script has changed:

running this script with the current master returns this error:

```
run ocp_nlp.py
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
/home/sot7rng/catkin_ws/src/spencer_mpc/acados_toolkit/examples/python/ocp_nlp.py in <module>()
     13 # ODE Model
     14 step = 0.1
---> 15 nlp.set_model(ode_fun, step)
     16 
     17 # Cost function

/home/sot7rng/local/lib/acados.py in set_model(self, f, step)
    424 
    425     def set_model(self, f, step):
--> 426         return _acados.ocp_nlp_set_model(self, f, step)
    427 
    428     def set_cost(self, ls_cost):

RuntimeError: An ODE model should have 2 inputs: state and controls
```
setting `implicit=False` in line 10 resolves the isssue.

closed 2018-04-25 by dkouzoup