acados/acados #162: Code generation not working

schoelst, 2017-11-24:
The generated code is cluttered with `CASADI_SYMBOL_EXPORT` and thus I am unable to compile it.
I don't think this issue is specific to my machine, since also files in the repository are affected e.g. see `examples/c/chain_model/vde_adj_chain_nm7.c` or run `grep -r CASADI_SYMBOL_EXPORT` to get a full picture. The first occurrence in the repo I found is 9a5915a804af0739eee2babd8b2c14ae94d835f9

roversch, 2017-11-24:
Could you post this in the CasADi issues please? Then I'll close it here (it's not a problem caused by acados)

jgillis, 2017-11-24:
WHhen you code-generate with CasADi, you may pass an option `with_export` `false` to suppress the interjection of `CASADI_SYMBOL_EXPORT`.

roversch, 2017-11-24:
@jgillis why is `CASADI_SYMBOL_EXPORT` not defined in the header anyway?

jgillis, 2017-11-24:
https://github.com/casadi/casadi/issues/2135

roversch, 2017-11-24:
Perfect! Closing this for now

closed 2017-11-24 by roversch