acados/acados #163: casadiMEX.mex: undefined symbol

rienq, 2017-11-24:
This error currently appears when compiling acados with unit testing set to ON
`cmake -D SWIG_MATLAB=0 -D SWIG_PYTHON=0 -DUNIT_TESTS=ON ..
make install`

The error looks as follows:
`[ 88%] Generating data for numerical integration test
cd /home/quirynen/Software/acados/build/test/sim && octave-cli --no-gui --path /home/quirynen/Software/acados/test/sim/pendulum/casadi:/home/quirynen/Software/acados/external/casadi-octave-v3.2.2 /home/quirynen/Software/acados/test/sim/pendulum/casadi/generate_pendulum.m
error: sym: /home/quirynen/Software/acados/external/casadi-octave-v3.2.2/casadiMEX.mex: failed to load: /home/quirynen/Software/acados/external/casadi-octave-v3.2.2/casadiMEX.mex: undefined symbol: _ZNK6casadi8Callback7eval_sxEPPKNS_6SXElemEPPS1_PiS5_i
error: execution exception in /home/quirynen/Software/acados/test/sim/pendulum/casadi/generate_pendulum.m
cd /home/quirynen/Software/acados/build/test/sim && sed -i.bak -e 's/real_t\ a0/mem\ =\ 0;\ mem\ +=\ 0;\ w\ =\ 0;\ w\ +=\ 0;\ iw\ =\ 0;\ iw\ +=\ 0;\ \ real_t\ a0/g' /home/quirynen/Software/acados/build/test/sim/jac_pendulum.c /home/quirynen/Software/acados/build/test/sim/vde_forw_pendulum.c /home/quirynen/Software/acados/build/test/sim/vde_adj_pendulum.c /home/quirynen/Software/acados/build/test/sim/vde_hess_pendulum.c
sed: can't read /home/quirynen/Software/acados/build/test/sim/jac_pendulum.c: No such file or directory
sed: can't read /home/quirynen/Software/acados/build/test/sim/vde_forw_pendulum.c: No such file or directory
sed: can't read /home/quirynen/Software/acados/build/test/sim/vde_adj_pendulum.c: No such file or directory
sed: can't read /home/quirynen/Software/acados/build/test/sim/vde_hess_pendulum.c: No such file or directory
test/sim/CMakeFiles/sim_gen.dir/build.make:64: recipe for target 'test/sim/jac_pendulum.c' failed`

I tried both setting the flag -D_GLIBCXX_USE_CXX11_ABI=1 and -D_GLIBCXX_USE_CXX11_ABI=0 in cmake but that did not change anything. Maybe something that is relevant, is that I currently cannot compile with SWIG either (which is currently not a priority for me, but it might be related to this issue maybe). Namely, if I call cmake with 
`cmake -D SWIG_MATLAB=1 -D SWIG_PYTHON=1 -DUNIT_TESTS=ON ..`
then cmake returns 
`-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type is Release
-- Installation directory is /home/quirynen/local
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Found FORTRANLIBS: /usr/lib/gcc/x86_64-linux-gnu/5/libgfortran.so  
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- MATLAB executable found: /usr/local/bin/matlab
-- Will look for MATLAB libraries in /usr/local/MATLAB/R2016b/bin/glnxa64
-- Found MATLAB: /usr/local/MATLAB/R2016b  
-- Found SWIG: /home/quirynen/Software/acados/external/swig/swig (found version "3.0.11") 
CMake Error at cmake/FindCasadiMatlab.cmake:49 (file):
  file failed to open for reading (No such file or directory):
    /usr/local/include/casadi/config.h
Call Stack (most recent call first):
  swig/matlab/CMakeLists.txt:2 (find_package)
`

roversch, 2017-11-24:
Looking at

```error: sym: /home/quirynen/Software/acados/external/casadi-octave-v3.2.2/casadiMEX.mex: failed to load: /home/quirynen/Software/acados/external/casadi-octave-v3.2.2/casadiMEX.mex: undefined symbol: _ZNK6casadi8Callback7eval_sxEPPKNS_6SXElemEPPS1_PiS5_i error: execution exception```

, could this be an error in your CasADi installation? Can you run CasADi 'stand-alone'?

rienq, 2017-11-24:
I can use CasADi from MATLAB without problems, I only have seen this issue when the acados CMake script wants to specifically run casadi-octave-v3.2.2 in order to generate the C files to run the unit tests

zanellia, 2017-11-24:
I still think this might be an ABI problem. Could you check which gcc
version has been used to compile your Python  binaries?

rienq, 2017-11-24:
I am using 5.4.0 myself, as the CMake output confirms, but I guess you are wondering about the version that is used to generate the CasADi binaries? I have no idea..

roversch, 2017-11-24:
Bear in mind that gcc 5.4.0 is not compatible with matlab (only gcc 4.9
is).. not sure about octave

rienq, 2017-11-29:
Both issues have been fixed by @roversch :
- undefined symbol error:
`export LD_LIBRARY_PATH=path_to_acados/external/casadi-octave-v3.2.3:$LD_LIBRARY_PATH`
- casadi/config.h not found:
The install.sh is outdated on the remote rienq/acados so it works, after removing all old installed versions of CasADi, and using the install.sh from
https://raw.githubusercontent.com/acados/acados/master/install.sh

closed 2017-11-29 by rienq