acados/acados #187: Issue with cmake's find MATLAB

dkouzoup, 2017-12-15:
I had recently issues on linux with finding the MATLAB installation.

1. I think it is not a good idea to overwrite Matlab_ROOT_DIR in case it is passed by the user. See:

- https://github.com/acados/acados/blob/master/swig/matlab/CMakeLists.txt#L1

2. What is exactly MAT_LIBRARY in the line below? It seems that it does not exist on the laptop I tested and after removing this requirement, the build is succesful. Do we need it?

- https://github.com/acados/acados/blob/master/swig/matlab/CMakeLists.txt#L2

dkouzoup, 2017-12-15:
@roversch

doanminhdang, 2017-12-15:
If it is the MATLAB R2017b, the latest cmake has a fix in finding MATLAB.
https://cmake.org/cmake/help/v3.10/module/FindMatlab.html

dkouzoup, 2017-12-18:
Hey @doanminhdang, I've seen this page, but which fix exactly are you referring to?

doanminhdang, 2017-12-18:
I just downloaded cmake 3.10 and used it, didn't check how they fix it.

dkouzoup, 2017-12-18:
Ok thanks for reporting. The issue about not overwriting the user defined path and my question about whether we need MAT_LIBRARY remain however ;)

roversch, 2018-04-24:
Hey Dimitris, the MAT library is for reading/writing to `.mat` files: https://de.mathworks.com/help/matlab/matlab_external/mat-file-library-and-include-files.html 

It seems we don't need it, I will remove it now.

As for the Matlab_ROOT_DIR: it's not overwritten, I think: 
https://github.com/acados/acados/blob/master/cmake/FindMatlab.cmake#L1106

FreyJo, 2019-08-07:
https://github.com/acados/acados/pull/419

closed 2019-08-07 by FreyJo