acados/acados #188: qpDUNES problems in new_core

bnovoselnik, 2017-12-18:
There are two main issues with qpDUNES that I've not been able to figure out so far.

Is there a way to properly cold start it?
In mass spring examples a low quality solution in acquired (max KKT residual is around 1e-2)...

bnovoselnik, 2017-12-18:
That's correct, the termination depends on stationarity tolerance. Unfortunately, nothing happens when you set it to 1e-12 (the default is 1e-6).

dkouzoup, 2017-12-18:
hmm, which is the KKT residual that is so high?

bnovoselnik, 2017-12-18:
It's res_g, I think that's the stationarity residual.

dkouzoup, 2017-12-18:
and can you check how many iterations does qpDUNES do on this example the first time the problem is solved?

bnovoselnik, 2017-12-18:
this is from mass_spring_qpdunes:

inf norm res: 3.253027e-02, 8.881784e-16, 4.440892e-16, 1.144447e-15

***************************************************************
total time             =         4.610 ms      =        100.00 %
condensing time =         0.000 ms      =          0.00 %
QP time              =         4.601 ms      =         99.80 %
interface time     =         0.009 ms      =          0.20 %
misc                   =         0.000 ms      =          0.00 %
QP iter               =                6
***************************************************************

dkouzoup, 2017-12-18:
sanity check: to check that our choice of stationarity tolerance reaches qpDUNES, set it to a negative value. Should do max number of iterations in that case. Does it? Also, if you set NREP to 2, how many iterations are needed in the second (warm-started) run? Just 1 or more?

bnovoselnik, 2017-12-18:
Well if I put it to negative value then it fails to solve it.

With NREP 2 this is the output:

inf norm res: 5.329071e-15, 8.881784e-15, 4.440892e-16, 1.145113e-15

total time      =         0.276 ms      =        100.00 %
condensing time =         0.000 ms      =          0.00 %
QP time         =         0.268 ms      =         97.10 %
interface time  =         0.007 ms      =          2.54 %
misc            =         0.001 ms      =          0.36 %
QP iter         =             1

dkouzoup, 2017-12-18:
What do you mean it fails to solve it? It should run e.g. 100 iterations and give a maximum iterations return value (which btw we also have in acados but not properly implemented) and the solution of the QP should have high accuracy, no?

bnovoselnik, 2017-12-18:
Well, I mean exactly what I said :)
If I put the tolerance to -1 I get the following message:

qpDUNES failed to solve the QP. Error code: 9

dkouzoup, 2017-12-18:
I guess we could bypass this error but never mind. It means at least that indeed when you change the stationarity tolerance something happens. I will look into this weird issue at another point, thanks for reporting back!

bnovoselnik, 2017-12-18:
Ah, yes. Now I see that this message is from the ACADOS interface and not from the qpDUNES itself. Yes, if I bypass this the solution is of high accuracy (after maximum of 100 iterations):

inf norm res: 1.953993e-14, 8.881784e-16, 4.440892e-16, 1.145113e-15

total time      =        14.635 ms      =        100.00 %
condensing time =         0.000 ms      =          0.00 %
QP time         =        14.628 ms      =         99.95 %
interface time  =         0.007 ms      =          0.05 %
misc            =         0.000 ms      =          0.00 %
QP iter         =           100

dkouzoup, 2018-04-25:
update: cold start is implemented. Somehow, qpDUNES stops one iteration before convergence yielding high KKT residuals. Current HACK to pass unit test: qpDUNES is called twice in a row (with warm-start) to return correct solution. To be investigated..