acados/acados #191: [new core] wrong last multiplier in HPMPC

dkouzoup, 2018-01-10:
There is a bug in the last multiplier returned by HPMPC, which results in high KKT tolerances for the mass spring example.

zanellia, 2018-01-10:
the multipliers for the last stage are associated with terminal equalities cast as inequalities. For some reason HPMPC returns different values.

zanellia, 2018-01-10:
It looks like changing the initial value for mu0 the same multipliers are returned by both HPIPM and HPMPC. Indeed the trick of adding bounds with the same value to introduce additional equalities might break the numerics?

dkouzoup, 2018-01-11:
out of curiosity: does this create the same problem for the constraint on x0?

zanellia, 2018-01-11:
If implemented properly, it should not, I guess: the constraint on x0 can be eliminated before solving the QP. If kept, and cast using a pair of coinciding lower/upper bounds it might cause the same ill-conditioning ? We should double check with @giaf.

dkouzoup, 2018-01-11:
True. You can comment in/out the ELIMINATE_X0 flag to check that. But I guess the same problem will arise there too.

giaf, 2018-01-11:
yep same problem there

giaf, 2018-01-11:
until I have time to implement internally the elimination of x0

dkouzoup, 2018-01-11:
the same problem holds for the constraint on the last state @giaf ;)

dkouzoup, 2018-04-25:
small update: in the currently implemented unit test, the KKT tolerance of HPMPC is high only when combined with partial condensing. If N2 = N, the KKT is small, even without x0 eliminated (I will check what happens if we eliminate it)

FreyJo, 2020-10-07:
There will not be further effort in maintaining HPMPC interface, since a) it is not maintained anymore and b) HPIPM implements the same algorithms but is better maintained, tested and more modular.

closed 2020-10-07 by FreyJo