acados/acados #196: Various bugs due to making assumptions on HPIPM core data layout

giaf, 2018-01-21:
Hey  guys,

I've spent the weekend fixing bugs around acados, which are triggered by the use of BLASFEO HP version.

All bugs boil down to making assumption about the way HPIPM internally stored its variables, e.g. the (wrong) assumptions that the slacks or the residuals are stored contiguously one stage after the other.

Such assumptions can not be made, and the variables in the QP structures should be accessed using the QP structure interface (based on BLASFEO structures) and not the underlying memory arrays.

roversch, 2018-01-21:
Can you link some examples of the bugs here? Just for our education.

roversch, 2018-01-21:
> #if 1
>
> double tmp;
>
> res[0] = 0.0;
> for (ii=0; ii<=N; ii++)
> {
> 	blasfeo_dvecnrm_inf(nx[ii]+nu[ii]+2*ns[ii], &qp_res->res_g[ii], 0, &tmp);
> 	res[0] = tmp > res[0] ? tmp : res[0];
> }
>
> res[1] = 0.0;
> for (ii=0; ii<N; ii++)
> {
> 	blasfeo_dvecnrm_inf(nx[ii+1], &qp_res->res_b[ii], 0, &tmp);
> 	res[1] = tmp > res[1] ? tmp : res[1];
> }
>
> res[2] = 0.0;
> for (ii=0; ii<=N; ii++)
> {
> 	blasfeo_dvecnrm_inf(2*nb[ii]+2*ng[ii]+2*ns[ii], &qp_res->res_d[ii], 0, &tmp);
> 	res[2] = tmp > res[2] ? tmp : res[2];
> }
>
> res[3] = 0.0;
> for (ii=0; ii<=N; ii++)
> {
> 	blasfeo_dvecnrm_inf(2*nb[ii]+2*ng[ii]+2*ns[ii], &qp_res->res_m[ii], 0, &tmp);
> 	res[3] = tmp > res[3] ? tmp : res[3];
> }
>
> #else
>
> // XXX this should be avoided, since it does employ strucutre of the HPIPM core that may change !!!
>
> // compute size of res_q, res_b, res_d and res_m
> int nvt = 0;
> int net = 0;
> int nct = 0;
>
> for(ii=0; ii<N; ii++)
> {
> 	nvt += nx[ii]+nu[ii]+2*ns[ii];
> 	net += nx[ii+1];
> 	nct += 2*nb[ii]+2*ng[ii]+2*ns[ii];
> }
> nvt += nx[ii]+nu[ii]+2*ns[ii];
> nct += 2*nb[ii]+2*ng[ii]+2*ns[ii];
>
> // compute infinity norms of residuals
> blasfeo_dvecnrm_inf(nvt, qp_res->res_g, 0, &res[0]);
> blasfeo_dvecnrm_inf(net, qp_res->res_b, 0, &res[1]);
> blasfeo_dvecnrm_inf(nct, qp_res->res_d, 0, &res[2]);
> blasfeo_dvecnrm_inf(nct, qp_res->res_m, 0, &res[3]);
>
> #endif

what about calling

blasfeo_pack_tran_dmat

directly on BAbt? Is this subject to change? If so, maybe we could provide
setters/getters for each field separately, instead of converting a full
ocp_qp each time..

Side question: is the ordering in lb = [lbx, lbu] or lb = [lbu, lbx] ?

giaf, 2018-01-21:
I didn't get the first question about BAbt...

My point here is: using the HPIPM interface is safe, making assumptions about where matrices and vectors are mapped internally in memory as arrays in HPIPM is not, as it can change with the BLASFEO LA version and/or over time.
E.g., in the piece of code above, for BLASFEO RF it works (since there is no padding in vectors), for BLASFEO HP it does not (since there is padding, and then res_g[0].pa+(nx[0]+nu[0]+2*ns[0]) != res_g[1].pa if nx[0]+nu[0]+2*ns[0] is not a multiple of 4).

About the second question, the order is [lbu, lbx]

roversch, 2018-01-21:
I guess we have to extend the HPIPM interface then, to allow setting single
matrices/vectors.

Now, there is only row major OCP -> ocp_qp and column major OCP -> ocp_qp.

giaf, 2018-01-21:
Yep the right way to do it is to extend HPIPM interface to allow that. When needed, we can do it.

roversch, 2018-01-21:
I can work on that, I need it for the C++ interface anyway.

dkouzoup, 2018-01-22:
Hey @giaf, thanks for the heads-up. I hope I didn't do that mistake so far ;)

closed 2018-03-29 by giaf