acados/acados #204: External Functions

nielsvd, 2018-02-06:
Continuation of the discussion held in PR #198.

In several parts of ``acados`` we need to be able to use what we call ``external_function``s, more specifically we typically need to evaluate expressions generated by ``casadi`` as c-code. Since ``casadi`` generated c-code in general needs an externally allocated workspace, the implementation becomes non-trivial. Both @giaf and I have started to implement this concept with slightly different philosophies in mind. Consider an OCP NLP. In principle we both pursue the aim to define as much as possible of the NLP problem in the ``ocp_nlp_in``-struct. However, another aim of the ``acados`` code is to assign the responsibility over the memory requirements of submodules to the parent module, recursively. As opening this issue suggests, there is a slight conflict here, strictly related to the external functions.

Let me try to make a short summary of the two approaches, feel free to make comments and to further specify the approaches.
* The approach of @giaf is to be strict about the definition of the entire NLP problem in the ``ocp_nlp_in`` struct. Since ``external_function``s require their own workspace/memory in general, but will not be known at construction of the nlp-solver, this needs to be allocated elsewhere. Relevant ``external_function``s are passed to the submodules (e.g., integrator) via their respective input structs.
* The approach of @nielsvd is to relax the the aim to define the entire NLP problem in the ``ocp_nlp_in`` struct. Let the memory and workspace of the ``external_function``s be allocated by the modules/submodules that need it. This means that they need to be known at construction time of the modules, which is implemented by defining the relevant function pointers in the respective ``args``-structs.

Since both of us are working on prototypes, I suggest we make a decision how to proceed based on that. My approach is implemented in PR #198. Can you point to your implementation @giaf?

roversch, 2018-02-06:
Let me point out one more pro of each approach:

- @giaf 's approach is more flexible, because adding a new AD tool does not require writing a new module. In this sense, external functions are regarded as a special case and not as a solver.
- @nielsvd 's approach is more in line with the rest of the code base, where allocation of the functions is done by the 'parent' solvers, similar as for `sim` and `ocp_qp`

nielsvd, 2018-02-15:
Technical discussion not needed.

closed 2018-02-15 by nielsvd